psvanstrom/esphome-p1reader #3: Reading from UART timed out at byte 0!

steeriks, 2020-12-16:
I am getting this error:

Aidon meter with Rj12 (activated) NodeMcu V3. 

[17:31:09][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:09][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:09][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:09][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:09][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:10][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:10][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:10][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:10][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:10][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:19][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:19][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:19][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:19][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:19][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:20][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:20][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:20][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:20][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:20][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:29][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:29][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:29][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:29][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:29][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:30][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:30][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:30][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:30][E][uart_esp8266:167]: Reading from UART timed out at byte 0!
[17:31:30][E][uart_esp8266:167]: Reading from UART timed out at byte 0!

maccan-tech, 2021-01-01:
Same here.
Tekniskaverken, Aidon 6442SE meter with Rj12 (activated).
Using ESP8266 (wemos d1).

forsberg, 2021-01-17:
Same here, also on Tekniska Verken with Aidon. Reason seems to be that Tekniska verken have adopted "Branschrekommendation 1.2", which is a binary protocol.

Protocol information available here: https://www.tekniskaverken.se/siteassets/tekniska-verken/elnat/aidonfd-rj12-han-interface-se-v13a.cleaned.pdf

Some more information here: https://github.com/aviborg/swedishP1

pp-svanstrom, 2021-01-18:
@forsberg Great find! From version 1.3 of the specification they changed to an ASCII format, which is what my code is expecting. I will take a look at the binary format and see if I can add a field for the user to control which version the meter is using.

forsberg, 2021-01-18:
I contacted Tekniska Verken kundsupport, and they are "painfully aware" and are "having a dialouge with their provider" about how to fix this. 

I asked them if they have a timeline. Hopefully this will be fixed by itself in a couple of months.

bortek, 2021-02-05:
I am also seeing this in my logs sporadically. My mete ris Landis E360. But I have more issue with CRC checksum which I put in a separate issue.

forsberg, 2021-02-14:
I have the beginnings of a fix for this [in a branch](https://github.com/forsberg/esphome-p1reader/tree/branschrekommendation-1.2-tekniska-verken), but it needs a bit of cleanup - checking CRC and handling HDLC flag bytes inside frame. 

Has been running stable for about a week on an Aidon 6442SE from Tekniska Verken, Linköping.

pp-svanstrom, 2021-02-19:
Awesome @forsberg ! You're very welcome to do a PR once you have finished the cleanup, great work 👍

maccan-tech, 2021-02-24:
@forsberg Your branch has works great for me now for a week.
But i get this every time it reads the data.
`[10:41:09][D][hdlc:479]: Expect 0x01 (array tag), got 01
...`

jpduhen, 2021-12-17:
https://github.com/esphome/issues/issues/815#issuecomment-733584340
It helped me to reset the MH-Z19B by holdoing the HD 7sec to GND and resetting the ESP. You have to calibrate the MH by keeping it in fresh air for 20 mins.