psvanstrom/esphome-p1reader #10: ESP8266 I/O 5V restistant

HenkeGoransson, 2021-05-12:
is the NPN-transistor really needed?
Shouldn't it be possible to connect TX directly to RX without level shifting?
Would make the entire wireing alot easier.

bortek, 2021-05-14:
I am not sure if you can run w/o it. Why dont you remove it and test without it?

Anteus, 2021-05-14:
Searched a bit on that too. And some report it is fine and some say it's speced at 3.3V and might fry the microcontroller but still reported it working with 5V for an extended period. But might decrease lifespan of it.

HenkeGoransson, 2021-05-17:
I have a Wemos mini D1 which I plan to test this on. Can get back to you when i've tested.

weigu1, 2021-07-22:
Hi,
Normally Rx is open collector. So if you use a pull-up (10k) to 3.3V, you have only 3.3V. The serial signal is inverted, but the newer core software of the ESPs is capable to invert the signal!

Serial.begin(115200, SERIAL_8N1, SERIAL_FULL, 1, true); // true inverts the signal

You find the circuit on:
http://weigu.lu/microcontroller/smartyReader_P1/index.html

blackcult, 2021-10-19:
Hello,
Were you successful with removing the NPN?

jagheterfredrik, 2022-02-13:
I tried with an ESP32, it worked fine with inverting there instead of using the NPN. I tried using the internal pullup resistor but I don’t think it’s large enough so I added a 10k resistor to 3.3V. I did have to use additional capacitors (2x10uF) to get the ESP32 to power correctly.

If the ESP8266 can power fine without additional caps, I think all you need is a 10k resistor.

robinelfving, 2022-09-18:
Does this mean that i can connect D1 like this:

5V and RTS to 5V.
Data GND and GND to GND.
TX to RX and invert the pin in code.

In other words exclude the voltage divider and NPN-transistor?

jagheterfredrik, 2022-09-19:
No transistor nor divider needed. Pullup resistor and a capacitor (if the energy meter doesn’t supply enough current)

robinelfving, 2022-09-19:
Thank you Fredrik!

Anteus, 2022-11-11:
For those who want to do the same to invert the pin in software just change uart to below. Might need to change GPIO ports if not on NodeMCU ESP8266.

uart:
  id: uart_bus
  tx_pin:
    number: GPIO1
    inverted: true
  rx_pin:
    number: GPIO3
    inverted: true
  baud_rate: 115200