psvanstrom/esphome-p1reader #11: Home Assistant configuration

dessibelle, 2021-06-17:
I've been running Home Assistant for some time, but am new to ESPHome and I've ran into some issues with the configuration before flashing my NodeMCU. I understand this is essentially a Home Assistant and/or ESPHome question, but I have not been able to find anything useful when googling, so attempting a post here.

It's the value for `hass_api_password` in particular I'm having problems understanding. I assume it is an API password for some REST API that can be enabled within home assistant. Is that at all correct? I've tried both [the `legacy_api_password` method](https://www.home-assistant.io/docs/authentication/providers/) and [the newer REST API](https://developers.home-assistant.io/docs/api/rest/) without being able to successfully connect (using curl that is, not `esphome-p1reader` which I haven't even attempted yet).

I've also considered that `hass_api_password` could be something ESPHome related, which I have very little knowledge about, but my research has come up blank there as well. I'd greatly appreciate if someone could point me in the right direction with regards to what `hass_api_password` is and how I can set it up. Thanks!

psvanstrom, 2021-08-06:
The `hass_api_password` value is the password that will be enabled for the ESPHome API that Home Assistant uses for communication. Just invent a password and replace `hass_api_password` with that value, then when you set up the integration in Home Assistant it will ask your for your API password and use that to communicate with your device.

closed 2021-08-08 by psvanstrom

dessibelle, 2021-08-10:
Great, thanks for clearing that out! 🙏