psvanstrom/esphome-p1reader #12: Sensor does not appear in new (2021.8) Energy dashboard?

fredrikbaberg, 2021-08-04:
I'm unable to find the sensor in the Energy tab.

I'm referred to: https://www.home-assistant.io/more-info/statistics/.

Another implementation is referenced in the release notes, https://github.com/zuidwijk/dsmr, but I have not tried that as I haven't verified if the pin mapping is the same.

fredrikbaberg, 2021-08-04:
I've done some experimentation based on ESPHome documentation and the `dsmr` repository. I'm still waiting for data to be visible in the Energy panel (can take up to 2 hours), but adding the following to `Cumulative Active Import` at least allows it to be added:
```
state_class: "measurement"
device_class: "energy"
last_reset_type: "never"
```

Emlben, 2021-08-05:
I have the same issue.
Was told to add, in addition to device class and state, this :
last_reset; "1970-01-01T00:00:00+00:00"

Same state, can add the cumulative active import but no data is showing.

fredrikbaberg, 2021-08-05:
> I have the same issue.
> Was told to add, in addition to device class and state, this :
> last_reset; "1970-01-01T00:00:00+00:00"
> 
> Same state, can add the cumulative active import but no data is showing.

I read that `last_reset_type: "never"` should set `last_reset` to epoch, so it should lead to the same result. There is now a pull request that allows the sensor to be added.

grobius, 2021-08-05:
I must be doing something wrong. Keep getting this error regardless of how I format the yaml

      [state_class] is an invalid option for [0]. Did you mean [state_topic], [device_class]?
      state_class: measurement [source p1reader.yaml:74]
      device_class: energy

      [last_reset_type] is an invalid option for [0]. Please check the indentation.
      last_reset_type: never [source p1reader.yaml:76]

fredrikbaberg, 2021-08-05:
@grobius have you formatted the lines in the same way as the pull request I've linked to this issue? It seems to complain about indentation.

grobius, 2021-08-05:
Yes, I did copy it from the pull request. Here is the full yaml:

esphome:
  name: electricity_meter
  platform: ESP8266
  board: nodemcu
  includes:
    - p1reader.h

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "electricity_meter"
    password: !secret fallback_password

captive_portal:

# Enable logging
logger:
  level: DEBUG
  baud_rate: 0 # disable logging over uart
  
# Enable Home Assistant API
api:
  password: !secret hass_api_password

ota:
  password: !secret ota_password

uart:
  id: uart_bus
  tx_pin: TX
  rx_pin: RX
  baud_rate: 115200
  
sensor:
- platform: custom
  lambda: |-
    auto meter_sensor = new P1Reader(id(uart_bus));
    App.register_component(meter_sensor);
    return {
      meter_sensor->cumulativeActiveImport,
      meter_sensor->cumulativeActiveExport,
      meter_sensor->cumulativeReactiveImport,
      meter_sensor->cumulativeReactiveExport,
      meter_sensor->momentaryActiveImport,
      meter_sensor->momentaryActiveExport,
      meter_sensor->momentaryReactiveImport,
      meter_sensor->momentaryReactiveExport,
      meter_sensor->momentaryActiveImportL1,
      meter_sensor->momentaryActiveExportL1,
      meter_sensor->momentaryActiveImportL2,
      meter_sensor->momentaryActiveExportL2,
      meter_sensor->momentaryActiveImportL3,
      meter_sensor->momentaryActiveExportL3,
      meter_sensor->momentaryReactiveImportL1,
      meter_sensor->momentaryReactiveExportL1,
      meter_sensor->momentaryReactiveImportL2,
      meter_sensor->momentaryReactiveExportL2,
      meter_sensor->momentaryReactiveImportL3,
      meter_sensor->momentaryReactiveExportL3,
      meter_sensor->voltageL1,
      meter_sensor->voltageL2,
      meter_sensor->voltageL3,
      meter_sensor->currentL1,
      meter_sensor->currentL2,
      meter_sensor->currentL3
    };
  sensors:
  - name: "Cumulative Active Import"
    unit_of_measurement: kWh
    accuracy_decimals: 3
    state_class: "measurement"
    device_class: "energy"
    last_reset_type: "never"
  - name: "Cumulative Active Export"
    unit_of_measurement: kWh
    accuracy_decimals: 3
  - name: "Cumulative Reactive Import"
    unit_of_measurement: kvarh
    accuracy_decimals: 3
  - name: "Cumulative Reactive Export"
    unit_of_measurement: kvarh
    accuracy_decimals: 3
  - name: "Momentary Active Import"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Active Export"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Reactive Import"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Export"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Active Import Phase 1"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Active Export Phase 1"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Active Import Phase 2"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Active Export Phase 2"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Active Import Phase 3"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Active Export Phase 3"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Reactive Import Phase 1"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Export Phase 1"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Import Phase 2"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Export Phase 2"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Import Phase 3"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Export Phase 3"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Voltage Phase 1"
    unit_of_measurement: V
    accuracy_decimals: 3
  - name: "Voltage Phase 2"
    unit_of_measurement: V
    accuracy_decimals: 3
  - name: "Voltage Phase 3"
    unit_of_measurement: V
    accuracy_decimals: 3
  - name: "Current Phase 1"
    unit_of_measurement: A
    accuracy_decimals: 3
  - name: "Current Phase 2"
    unit_of_measurement: A
    accuracy_decimals: 3
  - name: "Current Phase 3"
    unit_of_measurement: A
    accuracy_decimals: 3

grobius, 2021-08-05:
Yes, I copied it from pullrequest. Here is the full yaml:

esphome:
  name: electricity_meter
  platform: ESP8266
  board: nodemcu
  includes:
    - p1reader.h

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "electricity_meter"
    password: !secret fallback_password

captive_portal:

# Enable logging
logger:
  level: DEBUG
  baud_rate: 0 # disable logging over uart
  
# Enable Home Assistant API
api:
  password: !secret hass_api_password

ota:
  password: !secret ota_password

uart:
  id: uart_bus
  tx_pin: TX
  rx_pin: RX
  baud_rate: 115200
  
sensor:
- platform: custom
  lambda: |-
    auto meter_sensor = new P1Reader(id(uart_bus));
    App.register_component(meter_sensor);
    return {
      meter_sensor->cumulativeActiveImport,
      meter_sensor->cumulativeActiveExport,
      meter_sensor->cumulativeReactiveImport,
      meter_sensor->cumulativeReactiveExport,
      meter_sensor->momentaryActiveImport,
      meter_sensor->momentaryActiveExport,
      meter_sensor->momentaryReactiveImport,
      meter_sensor->momentaryReactiveExport,
      meter_sensor->momentaryActiveImportL1,
      meter_sensor->momentaryActiveExportL1,
      meter_sensor->momentaryActiveImportL2,
      meter_sensor->momentaryActiveExportL2,
      meter_sensor->momentaryActiveImportL3,
      meter_sensor->momentaryActiveExportL3,
      meter_sensor->momentaryReactiveImportL1,
      meter_sensor->momentaryReactiveExportL1,
      meter_sensor->momentaryReactiveImportL2,
      meter_sensor->momentaryReactiveExportL2,
      meter_sensor->momentaryReactiveImportL3,
      meter_sensor->momentaryReactiveExportL3,
      meter_sensor->voltageL1,
      meter_sensor->voltageL2,
      meter_sensor->voltageL3,
      meter_sensor->currentL1,
      meter_sensor->currentL2,
      meter_sensor->currentL3
    };
  sensors:
  - name: "Cumulative Active Import"
    unit_of_measurement: kWh
    accuracy_decimals: 3
    state_class: "measurement"
    device_class: "energy"
    last_reset_type: "never"
  - name: "Cumulative Active Export"
    unit_of_measurement: kWh
    accuracy_decimals: 3
  - name: "Cumulative Reactive Import"
    unit_of_measurement: kvarh
    accuracy_decimals: 3
  - name: "Cumulative Reactive Export"
    unit_of_measurement: kvarh
    accuracy_decimals: 3
  - name: "Momentary Active Import"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Active Export"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Reactive Import"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Export"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Active Import Phase 1"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Active Export Phase 1"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Active Import Phase 2"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Active Export Phase 2"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Active Import Phase 3"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Active Export Phase 3"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Reactive Import Phase 1"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Export Phase 1"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Import Phase 2"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Export Phase 2"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Import Phase 3"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Export Phase 3"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Voltage Phase 1"
    unit_of_measurement: V
    accuracy_decimals: 3
  - name: "Voltage Phase 2"
    unit_of_measurement: V
    accuracy_decimals: 3
  - name: "Voltage Phase 3"
    unit_of_measurement: V
    accuracy_decimals: 3
  - name: "Current Phase 1"
    unit_of_measurement: A
    accuracy_decimals: 3
  - name: "Current Phase 2"
    unit_of_measurement: A
    accuracy_decimals: 3
  - name: "Current Phase 3"
    unit_of_measurement: A
    accuracy_decimals: 3

grobius, 2021-08-05:
Yes, I copied it from the pullrequest. Here is the full yaml:

`esphome:
  name: electricity_meter
  platform: ESP8266
  board: nodemcu
  includes:
    - p1reader.h

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "electricity_meter"
    password: !secret fallback_password

captive_portal:

# Enable logging
logger:
  level: DEBUG
  baud_rate: 0 # disable logging over uart
  
# Enable Home Assistant API
api:
  password: !secret hass_api_password

ota:
  password: !secret ota_password

uart:
  id: uart_bus
  tx_pin: TX
  rx_pin: RX
  baud_rate: 115200
  
sensor:
- platform: custom
  lambda: |-
    auto meter_sensor = new P1Reader(id(uart_bus));
    App.register_component(meter_sensor);
    return {
      meter_sensor->cumulativeActiveImport,
      meter_sensor->cumulativeActiveExport,
      meter_sensor->cumulativeReactiveImport,
      meter_sensor->cumulativeReactiveExport,
      meter_sensor->momentaryActiveImport,
      meter_sensor->momentaryActiveExport,
      meter_sensor->momentaryReactiveImport,
      meter_sensor->momentaryReactiveExport,
      meter_sensor->momentaryActiveImportL1,
      meter_sensor->momentaryActiveExportL1,
      meter_sensor->momentaryActiveImportL2,
      meter_sensor->momentaryActiveExportL2,
      meter_sensor->momentaryActiveImportL3,
      meter_sensor->momentaryActiveExportL3,
      meter_sensor->momentaryReactiveImportL1,
      meter_sensor->momentaryReactiveExportL1,
      meter_sensor->momentaryReactiveImportL2,
      meter_sensor->momentaryReactiveExportL2,
      meter_sensor->momentaryReactiveImportL3,
      meter_sensor->momentaryReactiveExportL3,
      meter_sensor->voltageL1,
      meter_sensor->voltageL2,
      meter_sensor->voltageL3,
      meter_sensor->currentL1,
      meter_sensor->currentL2,
      meter_sensor->currentL3
    };
  sensors:
  - name: "Cumulative Active Import"
    unit_of_measurement: kWh
    accuracy_decimals: 3
    state_class: "measurement"
    device_class: "energy"
    last_reset_type: "never"
  - name: "Cumulative Active Export"
    unit_of_measurement: kWh
    accuracy_decimals: 3
  - name: "Cumulative Reactive Import"
    unit_of_measurement: kvarh
    accuracy_decimals: 3
  - name: "Cumulative Reactive Export"
    unit_of_measurement: kvarh
    accuracy_decimals: 3
  - name: "Momentary Active Import"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Active Export"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Reactive Import"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Export"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Active Import Phase 1"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Active Export Phase 1"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Active Import Phase 2"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Active Export Phase 2"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Active Import Phase 3"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Active Export Phase 3"
    unit_of_measurement: kW
    accuracy_decimals: 3
  - name: "Momentary Reactive Import Phase 1"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Export Phase 1"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Import Phase 2"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Export Phase 2"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Import Phase 3"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Export Phase 3"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Voltage Phase 1"
    unit_of_measurement: V
    accuracy_decimals: 3
  - name: "Voltage Phase 2"
    unit_of_measurement: V
    accuracy_decimals: 3
  - name: "Voltage Phase 3"
    unit_of_measurement: V
    accuracy_decimals: 3
  - name: "Current Phase 1"
    unit_of_measurement: A
    accuracy_decimals: 3
  - name: "Current Phase 2"
    unit_of_measurement: A
    accuracy_decimals: 3
  - name: "Current Phase 3"
    unit_of_measurement: A
    accuracy_decimals: 3
`

grobius, 2021-08-05:
Yes, I did copy it from the pullrequest. I can't see that there is something wrong with it (but it's probably me):

```
  sensors:
  - name: "Cumulative Active Import"
    unit_of_measurement: kWh
    accuracy_decimals: 3
    state_class: "measurement"
    device_class: "energy"
    last_reset_type: "never"
```

grobius, 2021-08-05:
Solved it. Update of esphome was required (I was running v1.18, upgrade to v1.20.4 solved it)

Emlben, 2021-08-05:
> Yes, I did copy it from the pullrequest. I can't see that there is something wrong with it (but it's probably me):
> 
>

Same error message here when pasting in ESPhome.

Emlben, 2021-08-05:
> Solved it. Update of esphome was required (I was running v1.18, upgrade to v1.20.4 solved it)

Dooh, same issue here ;)

Thanks mates!

psvanstrom, 2021-08-06:
Fixed by @fredrikbaberg in PR https://github.com/psvanstrom/esphome-p1reader/pull/13 👍

closed 2021-08-06 by psvanstrom