psvanstrom/esphome-p1reader #14: PCB

Naesstrom, 2021-08-06:
So I got inspired with the new energy dashboard, found this repo when searching for alternatives to the dutch P1 system.

Redid your schematic to use a wemos d1 mini instead
[image]

And ofc a matching PCB
[image]

PCB is open for anyone to copy, download etc. here https://oshwlab.com/Naesstrom/esphome-p1reader

Now I just need to wait for the PCB's to design the 3d printed case :D

psvanstrom, 2021-08-06:
This is awesome @Naesstrom! Can I incorporate your PCB image and the link to the PCB into the README.md of this project?

Naesstrom, 2021-08-06:
absolutely @psvanstrom, I can update you once I get the PCB's to. I ordered a bunch so I can send you one!

psvanstrom, 2021-08-06:
Thanks @Naesstrom! I updated the README with your PCB design. I would love to try out the PCB once you get it delivered 🙏

TobbeWendt, 2021-08-17:
@Naesstrom  I am interested to by 1-2 boards from you if possible. I live in Sweden.

Naesstrom, 2021-08-18:
Got them today but the 6/6 contacts from electrokit that I had didn't fit. If you have ordered from somewhere else they might fit tough!
This are the ones that didn't fit: https://www.electrokit.com/en/product/modular-connector-6-6-pcb/
This is the model I used: https://lcsc.com/product-detail/Ethernet-Connectors-Modular-Connectors-RJ45-RJ11_CONNFLY-Elec-DS1133-S60APS_C77857.html

So either I have to source a new 6/6 contact or order new pcbs

Naesstrom, 2021-09-09:
[image]
Uploaded the STL's for the box here. Tight fit but looks nice!
https://www.thingiverse.com/thing:4961372

psvanstrom, 2021-09-09:
Awesome, I'll add it to the README 👍

closed 2021-12-14 by psvanstrom

SchneHa, 2023-03-18:
R1 and R2 has to be changed
The PCB should be improved by inserting a electrolytic cap of at least 1000µF/16V between 5V and GND and a jumper to cut RX to make it easy flashable if built in enclosure
The thing files of the enclosure doesn't have a lid for the large version that fits. It seems to be molded with the enclosure itself to get a totally closed enclosure.

SchneHa, 2023-04-23:
Regarding the thing files:
Here you can find files for a well fitting enclosure with lid: https://www.thingiverse.com/thing:5564796

The PCB still needs an improvement.

cenobitedk, 2024-04-07:
@Naesstrom Do you have any extra PCB you're selling?

SchneHa, 2024-04-07:
Hello Jeppe,sure, but at the moment I‘m not at home for the next 3 weeks.Do you need the PCB only but also other parts?Best regards,Hans

cenobitedk, 2024-04-07:
@SchneHa Cool, I need the PCB and the 6P6 connecter if you have it? I have loads of resistors etc. already. Thanks

SchneHa, 2024-04-07:
Sure, and I think you need the capacitor as well as it has to fit between the connectors.Where are you?

SchneHa, 2024-04-07:
I think it was a misunderstanding. I‘m talking about this device: github.com

cenobitedk, 2024-04-07:
Not sure what you are linking to, I only see "github.com" and no link.

SchneHa, 2024-04-07:
Sorry, chopped by mistake: https://github.com/SchneHa/P1-WiFi-Gateway

cenobitedk, 2024-04-07:
Ah, yeah, ok, that's not what I'm looking for :) thanks anyway