psvanstrom/esphome-p1reader #18: Trouble installing node

kakanbro, 2021-09-11:
Hey, ESPhome is all new to me. I use the addon in homeassistant. I copied the content from your p1reader.yaml pasted to my yaml file that was auto generated and created a new secrets.yaml with my data in the same directory (/config/esphome) but when I try to install I always get the same error log. What am I doing wrong?

Compiling /data/p1reader/.pioenvs/p1reader/src/main.cpp.o
/config/esphome/p1reader.yaml: In lambda function:
/config/esphome/p1reader.yaml:38:31: error: expected type-specifier before 'P1Reader'
     auto meter_sensor = new P1Reader(id(uart_bus));
                               ^
/config/esphome/p1reader.yaml:67:7: error: could not convert '{<expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::sensor::Sensor*>'
     };
       ^
/config/esphome/p1reader.yaml:68:3: warning: control reaches end of non-void function [-Wreturn-type]
   sensors:
   ^
*** [/data/p1reader/.pioenvs/p1reader/src/main.cpp.o] Error 1
========================== [FAILED] Took 1.97 seconds ==========================

PeterHolgersson, 2021-09-11:
Did you also copy the p1reader.h to the same directory in esphome?

kakanbro, 2021-09-11:
I had missed that part, I copied it now, unfortunately I still get the same error msg though

PeterHolgersson, 2021-09-12:
Please check the indentation as you have copied the text, especially the "Sensor:" part:
`sensor:
- platform: custom
  lambda: |-
    auto meter_sensor = new P1Reader(id(uart_bus));
    App.register_component(meter_sensor);
    return {
      meter_sensor->cumulativeActiveImport,
      meter_sensor->cumulativeActiveExport,
      meter_sensor->cumulativeReactiveImport,
      meter_sensor->cumulativeReactiveExport,
      meter_sensor->momentaryActiveImport,
      meter_sensor->momentaryActiveExport,
      meter_sensor->momentaryReactiveImport,
      meter_sensor->momentaryReactiveExport,
      meter_sensor->momentaryActiveImportL1,
      meter_sensor->momentaryActiveExportL1,
      meter_sensor->momentaryActiveImportL2,
      meter_sensor->momentaryActiveExportL2,
      meter_sensor->momentaryActiveImportL3,
      meter_sensor->momentaryActiveExportL3,
      meter_sensor->momentaryReactiveImportL1,
      meter_sensor->momentaryReactiveExportL1,
      meter_sensor->momentaryReactiveImportL2,
      meter_sensor->momentaryReactiveExportL2,
      meter_sensor->momentaryReactiveImportL3,
      meter_sensor->momentaryReactiveExportL3,
      meter_sensor->voltageL1,
      meter_sensor->voltageL2,
      meter_sensor->voltageL3,
      meter_sensor->currentL1,
      meter_sensor->currentL2,
      meter_sensor->currentL3
    };`

kakanbro, 2021-09-13:
I ended up deleting the yaml and copiying the entire file instead of copy the data within it. Must have been a indentation problem when I copy pasted. It is working as intented now, thanks :)

psvanstrom, 2021-09-17:
Great to hear that you solved it 👍 . I'll close the issue.

closed 2021-09-17 by psvanstrom