psvanstrom/esphome-p1reader #20: Compilation error with latest esphome 2021.10.0

newlund, 2021-10-21:
In file included from src/main.cpp:52:0:
src/p1reader.h: In member function 'void P1Reader::readP1Message()':
src/p1reader.h:261:58: error: 'readBytesUntil' was not declared in this scope
           int len = readBytesUntil('\n', buffer, BUF_SIZE);
                                                          ^
Compiling /data/energy/.pioenvs/energy/lib67b/ESP8266WiFi/CertStoreBearSSL.cpp.o
*** [/data/energy/.pioenvs/energy/src/main.cpp.o] Error 1
========================= [FAILED] Took 24.46 seconds =========================

pp-svanstrom, 2021-10-21:
I'm on it, just realised it myself :) Fix will be up shortly 👍

newlund, 2021-10-21:
Wow! That was fast! :)
Working fine now. Thanks! 👍

psvanstrom, 2021-10-21:
They just released 2021.10.1 which now failed to compile with one of the changes I had to do for version 2021.10.0 🙄 I guess it was something that wasn't supposed to go out.

Anyway, I've merged a new version that now works for version 2021.10.1.