psvanstrom/esphome-p1reader #23: Issue compiling

RealElwood, 2021-11-16:
I get the following when installing, i get 1 device and no entities.
Wired with the custom PCB and a wemos d1 mini.

`INFO Reading configuration /config/esphome/p1-reader.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing p1-reader (board: nodemcu; framework: arduino; platform: platformio/espressif8266 @ 2.6.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 2.0.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <noise-c> 0.1.4
|   |-- <libsodium> 1.10018.1
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling /data/p1-reader/.pioenvs/p1-reader/src/main.cpp.o
Compiling /data/p1-reader/.pioenvs/p1-reader/lib67b/ESP8266WiFi/WiFiClientSecureBearSSL.cpp.o
Compiling /data/p1-reader/.pioenvs/p1-reader/lib67b/ESP8266WiFi/WiFiServer.cpp.o
Compiling /data/p1-reader/.pioenvs/p1-reader/lib67b/ESP8266WiFi/WiFiServerSecureAxTLS.cpp.o
In file included from src/main.cpp:53:0:
src/p1reader.h:97:3: error: stray '\302' in program
   <title>esphome-p1reader/p1reader.h at main · psvanstrom/esphome-p1reader · GitHub</title>
   ^
src/p1reader.h:97:3: error: stray '\267' in program
src/p1reader.h:97:3: error: stray '\302' in program
src/p1reader.h:97:3: error: stray '\267' in program
In file included from src/main.cpp:53:0:
src/p1reader.h:366:10: warning: missing terminating ' character [enabled by default]
     </option></form><form class="js-site-search-form" role="search" aria-label="Site" data-scope-type="Repository" data-scope-id="310318870" data-scoped-search-url="/psvanstrom/esphome-p1reader/search" data-owner-scoped-search-url="/users/psvanstrom/search" data-unscoped-search-url="/search" action="/psvanstrom/esphome-p1reader/search" accept-charset="UTF-8" method="get">
          ^
src/p1reader.h:366:5: error: missing terminating ' character
     </option></form><form class="js-site-search-form" role="search" aria-label="Site" data-scope-type="Repository" data-scope-id="310318870" data-scoped-search-url="/psvanstrom/esphome-p1reader/search" data-owner-scoped-search-url="/users/psvanstrom/search" data-unscoped-search-url="/search" action="/psvanstrom/esphome-p1reader/search" accept-charset="UTF-8" method="get">
     ^
src/p1reader.h:423:7: error: stray '\342' in program
       <span aria-hidden="true" class="d-inline-block ml-1 v-align-middle">↵</span>
       ^
src/p1reader.h:423:7: error: stray '\206' in program
src/p1reader.h:423:7: error: stray '\265' in program
src/p1reader.h:428:7: error: stray '\342' in program
       <span class="d-inline-block ml-1 v-align-middle">↵</span>
       ^
src/p1reader.h:428:7: error: stray '\206' in program
src/p1reader.h:428:7: error: stray '\265' in program
src/p1reader.h:470:7: error: stray '\342' in program
       <span aria-hidden="true" class="d-inline-block ml-1 v-align-middle">↵</span>
       ^
src/p1reader.h:470:7: error: stray '\206' in program
src/p1reader.h:470:7: error: stray '\265' in program
src/p1reader.h:475:7: error: stray '\342' in program
       <span class="d-inline-block ml-1 v-align-middle">↵</span>
       ^
src/p1reader.h:475:7: error: stray '\206' in program
src/p1reader.h:475:7: error: stray '\265' in program
src/p1reader.h:508:7: error: stray '\342' in program
       <span aria-hidden="true" class="d-inline-block ml-1 v-align-middle">↵</span>
       ^
src/p1reader.h:508:7: error: stray '\206' in program
src/p1reader.h:508:7: error: stray '\265' in program
src/p1reader.h:513:7: error: stray '\342' in program
       <span class="d-inline-block ml-1 v-align-middle">↵</span>
       ^
[... 260 lines cut ...]
                       <li class="edge-item-fix"><a href="https://github.com/events" class="py-2 lh-condensed-ultra d-block Link--secondary no-underline f5 Bump-link--hover" data-ga-click="(Logged out) Header, go to Events">Events <span class="Bump-link-symbol float-right text-normal color-fg-muted pr-3">&rarr;</span></a></li>
                                                                                                                                                                                                                                                                                                                        ^
src/p1reader.h:319:329: error: expected unqualified-id before '<' token
                       <li class="edge-item-fix"><a href="https://github.community" class="py-2 lh-condensed-ultra d-block Link--secondary no-underline f5 Bump-link--hover" data-ga-click="(Logged out) Header, go to Community forum">Community forum <span class="Bump-link-symbol float-right text-normal color-fg-muted pr-3">&rarr;</span></a></li>
                                                                                                                                                                                                                                                                                                                                         ^
src/p1reader.h:320:335: error: expected unqualified-id before '<' token
                       <li class="edge-item-fix"><a href="https://education.github.com" class="py-2 lh-condensed-ultra d-block Link--secondary no-underline f5 Bump-link--hover" data-ga-click="(Logged out) Header, go to GitHub Education">GitHub Education <span class="Bump-link-symbol float-right text-normal color-fg-muted pr-3">&rarr;</span></a></li>
                                                                                                                                                                                                                                                                                                                                               ^
src/p1reader.h:321:344: error: expected unqualified-id before '<' token
                       <li class="edge-item-fix"><a href="https://stars.github.com" class="py-2 pb-0 lh-condensed-ultra d-block Link--secondary no-underline f5 Bump-link--hover" data-ga-click="(Logged out) Header, go to GitHub Stars Program">GitHub Stars program <span class="Bump-link-symbol float-right text-normal color-fg-muted pr-3">&rarr;</span></a></li>
                                                                                                                                                                                                                                                                                                                                                        ^
src/p1reader.h:341:265: error: expected unqualified-id before '<' token
                     <a href="/pricing" class="pb-2 lh-condensed-ultra d-block Link--primary no-underline h5 Bump-link--hover" data-ga-click="(Logged out) Header, go to Pricing">Plans <span class="Bump-link-symbol float-right text-normal color-fg-muted pr-3">&rarr;</span></a>
                                                                                                                                                                                                                                                                         ^
src/p1reader.h:344:328: error: expected unqualified-id before '<' token
                       <li class="edge-item-fix"><a href="/pricing#feature-comparison" class="py-2 lh-condensed-ultra d-block Link--secondary no-underline f5 Bump-link--hover" data-ga-click="(Logged out) Header, go to Compare plans">Compare plans <span class="Bump-link-symbol float-right text-normal color-fg-muted pr-3">&rarr;</span></a></li>
                                                                                                                                                                                                                                                                                                                                        ^
src/p1reader.h:345:338: error: expected unqualified-id before '<' token
                       <li class="edge-item-fix"><a href="https://github.com/enterprise/contact" class="py-2 lh-condensed-ultra d-block Link--secondary no-underline f5 Bump-link--hover" data-ga-click="(Logged out) Header, go to Contact Sales">Contact Sales <span class="Bump-li [... cut ...] /span></a></li>
                                                                                                                                                                                                                                                                                                                                                  ^
*** [/data/p1-reader/.pioenvs/p1-reader/src/main.cpp.o] Error 1
========================= [FAILED] Took 22.21 seconds =========================`

Anteus, 2021-11-16:
Check if you have the weird chars in the source code files p1reader.h and p1reader.yaml.
https://arduino.stackexchange.com/questions/30581/stray-302-in-program-error-when-compiling

RealElwood, 2021-11-16:
Well i deleted the p1reader.h and used the latest and now the installation went through. I must've DL'd an old version from the beginning. Even though it was long after the latest update.

RealElwood, 2021-11-17:
All above issues resolved, probably an unfortunate series of events. What worked was to uninstall ESPHome from Home Assistant. Delete the whole ESPHome folder using Samba, and also starting "Let's Encrypt". Not sure if the last part did anything. When ESPHome was installed all communication worked. Now i just need to verify that my power meter is active on the port and that if follows protocol 1.3 (It's a Kamstrup OmniPower meter with an HAN adapter)

closed 2021-11-17 by RealElwood