psvanstrom/esphome-p1reader #24: Missing data

johanvh83, 2021-12-04:
Thanks for your nice 'project'. I've build your esphome p1 reader and it seems it works because I receive data in the logs:

`[12:45:52][I][crc:276]: Telegram read. CRC: 5895 = 5895. PASS = YES
[12:45:52][D][sensor:113]: 'Cumulative Active Import': Sending state 0.00000 kWh with 3 decimals of accuracy
[12:45:52][D][sensor:113]: 'Cumulative Active Export': Sending state 0.00000 kWh with 3 decimals of accuracy
[12:45:52][D][sensor:113]: 'Cumulative Reactive Import': Sending state 0.00000 kvarh with 3 decimals of accuracy
[12:45:52][D][sensor:113]: 'Cumulative Reactive Export': Sending state 0.00000 kvarh with 3 decimals of accuracy
[12:45:52][D][sensor:113]: 'Momentary Active Import': Sending state 0.43500 kW with 3 decimals of accuracy
[12:45:52][D][sensor:113]: 'Momentary Active Export': Sending state 0.00000 kW with 3 decimals of accuracy
[12:45:52][D][sensor:113]: 'Momentary Reactive Import': Sending state 0.00000 kvar with 3 decimals of accuracy
[12:45:52][D][sensor:113]: 'Momentary Reactive Export': Sending state 0.00000 kvar with 3 decimals of accuracy
[12:45:52][D][sensor:113]: 'Momentary Active Import Phase 1': Sending state 0.43500 kW with 3 decimals of accuracy
[12:45:52][D][sensor:113]: 'Momentary Active Export Phase 1': Sending state 0.00000 kW with 3 decimals of accuracy
[12:45:52][D][sensor:113]: '': Sending state 0.00000  with 0 decimals of accuracy
[12:45:52][D][sensor:113]: '': Sending state 0.00000  with 0 decimals of accuracy
[12:45:52][D][sensor:113]: '': Sending state 0.00000  with 0 decimals of accuracy
[12:45:52][D][sensor:113]: '': Sending state 0.00000  with 0 decimals of accuracy
[12:45:52][D][sensor:113]: 'Momentary Reactive Import Phase 1': Sending state 0.00000 kvar with 3 decimals of accuracy
[12:45:52][D][sensor:113]: 'Momentary Reactive Export Phase 1': Sending state 0.00000 kvar with 3 decimals of accuracy
[12:45:52][D][sensor:113]: '': Sending state 0.00000  with 0 decimals of accuracy
[12:45:52][D][sensor:113]: '': Sending state 0.00000  with 0 decimals of accuracy
[12:45:52][D][sensor:113]: '': Sending state 0.00000  with 0 decimals of accuracy
[12:45:52][D][sensor:113]: '': Sending state 0.00000  with 0 decimals of accuracy
[12:45:52][D][sensor:113]: 'Voltage Phase 1': Sending state 232.80000 V with 3 decimals of accuracy
[12:45:52][D][sensor:113]: '': Sending state 0.00000  with 0 decimals of accuracy
[12:45:52][D][sensor:113]: '': Sending state 0.00000  with 0 decimals of accuracy
[12:45:52][D][sensor:113]: 'Current Phase 1': Sending state 3.66000 A with 3 decimals of accuracy
[12:45:52][D][sensor:113]: '': Sending state 0.00000  with 0 decimals of accuracy`

As you see most entries contain the value 0.00000 except Momentary Active Import, Voltage Phase 1 and Current Phase 1.
Any idea why the other information does not appear.
My smart meter is type

psvanstrom, 2021-12-14:
Interesting, could be a parse problem. I will take a look 👍

psvanstrom, 2021-12-14:
Ok, so checking the obis codes, this looks like a simple fix by just updating the code that parses the codes.

For example, you have two cumulative import / export values, one pair for rate 1 (1.8.1 / 2.8.1) and one for rate 2 (1.8.2 / 2.8.2) while the Swedish specification only have one rate (1.8.0 / 2.8.0).

The empty values such as `'': Sending state 0.00000 with 0 decimals of accuracy` are for the second and third phases that your meter doesn't have.

So what you can do is to define a couple of new sensors for rate 1 / rate 2 like this to the `ParsedMessage` class:
```
class ParsedMessage {
  public:
    double cumulativeActiveImportRate1;
    double cumulativeActiveImportRate2;
    double cumulativeActiveExportRate1;
    double cumulativeActiveExportRate2;
    ...
```
and to the P1Reader class:
```
  public:
    Sensor *cumulativeActiveImportRate1 = new Sensor();
    Sensor *cumulativeActiveImportRate2 = new Sensor();
    Sensor *cumulativeActiveExportRate1 = new Sensor();
    Sensor *cumulativeActiveExportRate2 = new Sensor();
 ```

update the parse logic:
```
void parseRow(ParsedMessage* parsed, char* obisCode, char* value) {
      if (strncmp(obisCode, "1.8.1", 5) == 0) {
        parsed->cumulativeActiveImportRate1 = atof(value);

      } else if (strncmp(obisCode, "1.8.2", 5) == 0) {
        parsed->cumulativeActiveImportRate2 = atof(value);

      } else if (strncmp(obisCode, "2.8.1", 5) == 0) {
        parsed->cumulativeActiveExportRate1 = atof(value);

      } else if (strncmp(obisCode, "2.8.2", 5) == 0) {
        parsed->cumulativeActiveExportRate2 = atof(value);
      ....
```

and finally add them to the yaml so that they are exposed as sensors:
```
sensor:
- platform: custom
  lambda: |-
    auto meter_sensor = new P1Reader(id(uart_bus));
    App.register_component(meter_sensor);
    return {
      meter_sensor->cumulativeActiveImportRate1,
      meter_sensor->cumulativeActiveExportRate2,
      meter_sensor->cumulativeActiveImportRate1,
      meter_sensor->cumulativeActiveExportRate2,
      ...
```

You could also remove all sensors and parse logic for the 2nd and 3rd phases as you don't have those: `momentaryActiveImportL2`, `momentaryActiveImportL3` etc.

closed 2022-01-09 by psvanstrom