psvanstrom/esphome-p1reader #26: Vattenfall S34U18 change

Spitfire69, 2021-12-08:
For the S23U18 model that Vattenfall is distributing now you have to increase the buffer slightly.
I tried with #define BUF_SIZE 60 and it seems sufficent.

psvanstrom, 2021-12-14:
Cool, I will add it to the main branch and update the README with info about the Sanxing SX631 meter 👍

Spitfire69, 2021-12-14:
Yes, please do that. I have been running mine fine since a week ago now with no issues whatsoever.

Slamkryp, 2022-01-18:
@Spitfire69 Hi! I have a few questions for you :-)

First of all, did you have any problems with getting this to work except from the buffer? I cant get it to work, i get the data from the meter but i get crc fail. What hardware are you using?

And iam interrested in what data you get from the meter as i dont get very much from my sanxing (dont have it in front of me right now) but i only think its like 8 lines and no voltages och individual phase data at all.

Looking forward for you answer :-)

Spitfire69, 2022-01-18:
Hi, I’m using a Wemos D1 mini and it has been working flawlessly for over a month now. As mentioned the only thing needed was to extend the buffer. However before I did that I got CRC errors so my suggestion is to just add more bytes and test. Good luck!

Slamkryp, 2022-01-19:
Ok, iam using D1 mini, tried increasing buffer to 100 (was 60, but i dont know what it was when you made yours) and still no luck with crc :-(

So for now i bypassed the crc check and now i get data to home assistant but the energy dashboard is still blank.. well well.. 

> Hi, I’m using a Wemos D1 mini and it has been working flawlessly for over a month now. As mentioned the only thing needed was to extend the buffer. However before I did that I got CRC errors so my suggestion is to just add more bytes and test. Good luck!

Slamkryp, 2022-01-19:
@Spitfire69 
What does your sanxing output?
this is what i get from mine, would like to compare your output with mine

[21:43:03][D][data:265]: /AUX5
[21:43:03][D][data:265]: 1-3:0.2.8(50)
[21:43:03][D][data:265]: 0-0:1.0.0(220119214300W)
[21:43:03][D][data:265]: 0-0:96.1.1(SANXING SMART ELECTRIC)
[21:43:03][D][data:265]: 1-0:1.8.1(031189.251*kWh)
[21:43:03][D][data:265]: 1-0:1.8.2(000000.000*kWh)
[21:43:03][D][data:265]: 1-0:2.8.1(000000.000*kWh)
[21:43:03][D][data:265]: 1-0:2.8.2(000000.000*kWh)
[21:43:03][D][data:265]: 0-0:96.14.0(RATE1)
[21:43:03][D][data:265]: 1-0:1.7.0(04.407*kW)
[21:43:03][D][data:265]: 1-0:2.7.0(00.000*kW)
[21:43:04][D][data:265]: !8B77
[21:43:04][I][crc:276]: Telegram read. CRC: E7C2 = 8B77. PASS = NO

Spitfire69, 2022-01-19:
Yours is not even close…has Vattenfall installed your or is it a standard Sanxing? If that is the case I guess they have different firmware.

Here’s my output:

[21:46:43][I][crc:276]: Telegram read. CRC: 84C3 = 84C3. PASS = YES
[21:46:43][D][sensor:113]: 'Cumulative Active Import': Sending state 1905.56995 kWh with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Cumulative Active Export': Sending state 6.24500 kWh with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Cumulative Reactive Import': Sending state 0.00100 kvarh with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Cumulative Reactive Export': Sending state 761.00598 kvarh with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Momentary Active Import': Sending state 2.46200 kW with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Momentary Active Export': Sending state 0.00000 kW with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Momentary Reactive Import': Sending state 0.00000 kvar with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Momentary Reactive Export': Sending state 0.74100 kvar with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Momentary Active Import Phase 1': Sending state 0.11300 kW with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Momentary Active Export Phase 1': Sending state 0.00000 kW with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Momentary Active Import Phase 2': Sending state 0.12500 kW with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Momentary Active Export Phase 2': Sending state 0.00000 kW with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Momentary Active Import Phase 3': Sending state 2.22300 kW with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Momentary Active Export Phase 3': Sending state 0.00000 kW with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Momentary Reactive Import Phase 1': Sending state 0.00000 kvar with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Momentary Reactive Export Phase 1': Sending state 0.16300 kvar with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Momentary Reactive Import Phase 2': Sending state 0.02600 kvar with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Momentary Reactive Export Phase 2': Sending state 0.00000 kvar with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Momentary Reactive Import Phase 3': Sending state 0.00000 kvar with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Momentary Reactive Export Phase 3': Sending state 0.60500 kvar with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Voltage Phase 1': Sending state 231.89999 V with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Voltage Phase 2': Sending state 232.00000 V with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Voltage Phase 3': Sending state 227.89999 V with 3 decimals of accuracy
[21:46:43][D][sensor:113]: 'Current Phase 1': Sending state 0.90000 A with 3 decimals of accuracy

Slamkryp, 2022-01-20:
@Spitfire69 Can you please paste the meter output instead of what asphome sends to home assistant? :-)