psvanstrom/esphome-p1reader #30: Add mqtt support

calania, 2021-12-27:
Hello! Would it be possible to add mqtt support for us that use something other than home assist?

psvanstrom, 2021-12-29:
ESPHome has an extensive support for MQTT so it is absolutely possible to configure the p1-reader to publish MQTT events instead of using the Home Assistant API. This would not require any changes to the code, only configuration changes in the yaml file: https://esphome.io/components/mqtt.html

calania, 2021-12-30:
Sounds awesome! Will have a look.

closed 2021-12-30 by calania

ulfwin, 2021-12-30:
@calania Please let us know your findings! I'm using openhab so I'm also very interested in this 😁

calania, 2022-01-05:
@ulfwin Sadly I haven't received my meter yet but I will report back once I get it!