psvanstrom/esphome-p1reader #32: corrupt data?

admanbj, 2022-01-08:
I am getting this error from Log

[image]
"ip address is removed"

INFO Reading configuration /config/esphome/p1-log.yaml...
INFO Starting log output from  using esphome API
INFO Successfully connected to 
[13:50:44][I][app:102]: ESPHome version 2021.12.3 compiled on Jan  8 2022, 13:38:45
[13:50:44][C][wifi:488]: WiFi:
[13:50:44][C][wifi:350]:   Local MAC: XX:XX:12:F5:A0:3F
[13:50:44][C][wifi:351]:   SSID: [redacted]
[13:50:44][C][wifi:352]:   IP Address: 
[13:50:44][C][wifi:354]:   BSSID: [redacted]
[13:50:44][C][wifi:355]:   Hostname: 'p1rj12'
[13:50:44][C][wifi:357]:   Signal strength: -72 dB ▂▄▆█
[13:50:44][C][wifi:361]:   Channel: 1
[13:50:44][C][wifi:362]:   Subnet: 
[13:50:44][C][wifi:363]:   Gateway: 
[13:50:44][C][wifi:364]:   DNS1: 
[13:50:44][C][wifi:365]:   DNS2: 0.0.0.0
[13:50:44][C][logger:233]: Logger:
[13:50:44][C][logger:234]:   Level: DEBUG
[13:50:44][C][logger:235]:   Log Baud Rate: 0
[13:50:44][C][logger:236]:   Hardware UART: UART0
[13:50:44][C][uart.arduino_esp8266:100]: UART Bus:
[13:50:44][C][uart.arduino_esp8266:101]:   TX Pin: GPIO1
[13:50:44][C][uart.arduino_esp8266:102]:   RX Pin: GPIO3
[13:50:44][C][uart.arduino_esp8266:104]:   RX Buffer Size: 256
[13:50:44][C][uart.arduino_esp8266:106]:   Baud Rate: 115200 baud
[13:50:44][C][uart.arduino_esp8266:107]:   Data Bits: 8
[13:50:44][C][uart.arduino_esp8266:108]:   Parity: NONE
[13:50:44][C][uart.arduino_esp8266:109]:   Stop bits: 1
[13:50:44][C][uart.arduino_esp8266:111]:   Using hardware serial interface.
[13:50:44][C][captive_portal:144]: Captive Portal:
[13:50:44][C][ota:082]: Over-The-Air Updates:
[13:50:44][C][ota:083]:   Address: 
[13:50:44][C][ota:086]:   Using Password.
[13:50:44][C][api:134]: API Server:
[13:50:44][C][api:135]:   Address: 
[13:50:44][C][api:139]:   Using noise encryption: NO
[13:50:44][C][mdns:084]: mDNS:
[13:50:44][C][mdns:085]:   Hostname: p1rj12
[13:50:50][D][data:265]: ~\xa2CA\x83\x85\xeb\xe6\xe7
[13:50:50][D][data:265]: 
[13:50:50][D][data:265]: 
[13:50:50][D][data:265]: \xff!	
[13:50:50][D][data:265]: #	
[13:50:50][D][data:265]: 
[13:50:50][D][data:265]: 
[13:50:50][D][data:265]: 
[13:50:50][D][data:265]: 
[13:50:51][D][data:265]: 
[13:51:00][D][data:265]: ~\xa2CA\x83\x85\xeb\xe6\xe7
[13:51:00][D][data:265]: 
[13:51:00][D][data:265]: 
[13:51:00][D][data:265]: \xff!	
[13:51:00][D][data:265]: #	
[13:51:00][D][data:265]: 
[13:51:00][D][data:265]: 
[13:51:00][D][data:265]: 
[13:51:00][D][data:265]: 
[13:51:01][D][data:265]: 
[13:51:10][D][data:265]: ~\xa2CA\x83\x85\xeb\xe6\xe7
[13:51:10][D][data:265]: \xff\x80
[13:51:10][D][data:265]: '
[13:51:10][D][data:265]: \xff!	
[13:51:10][D][data:265]: \xff#	
[13:51:10][D][data:265]: 
[13:51:10][D][data:265]: 6
[13:51:10][D][data:265]: 
[13:51:10][D][data:265]: 
[13:51:10][D][data:265]: 
[13:51:11][D][data:265]: ~
[13:51:20][D][data:265]: ~\xa2CA\x83\x85\xeb\xe6\xe7
[13:51:20][D][data:265]: 
[13:51:20][D][data:265]: 
[13:51:20][D][data:265]: \xff!	
[13:51:20][D][data:265]: #	
[13:51:20][D][data:265]: 
[13:51:20][D][data:265]: 
[13:51:20][D][data:265]: 
[13:51:20][D][data:265]: 
[13:51:21][D][data:265]: 

admanbj, 2022-01-08:
Aidon electricity meter!

closed 2022-01-08 by admanbj

dimme, 2022-01-09:
The Aidon doesn't transmit ASCII. It's another binary format.