psvanstrom/esphome-p1reader #34: Add documentation for ESP32

Nosen92, 2022-01-11:
The code works great on the [NodeMCU ESP32](https://esphome.io/devices/nodemcu_esp32.html), but because the documentation doesn't cover it, there are some pitfalls.

First, the NodeMCU ESP32 is too power hungry for the P1 port, which seemingly only outputs 5V 250mA, so an external power source is required. This means the VIN line can be omitted. (I'm fairly certain of this fact, but it's possible you can get away with just 250mA. I don't feel like finding out the hard way though.)

Second, the NodeMCU ESP32 has three separate UARTs, with different pin configurations. The RX pin of UART0 is pin 3, and this fact wasn't very easy to find. ESPHome documentation simply declares it "Internal".

**I therefore suggest, for the sake of all users of the NodeMCU ESP32, that the documentation is updated to support them.**

All this being said, because the ESP32 is both more expensive and requires an external power source, I recommend new users use a Wemos D1 Mini if possible.

tedenda, 2022-01-11:
According to [http://www.weigu.lu/microcontroller/smartyReader_P1/index.html#link_3](http://www.weigu.lu/microcontroller/smartyReader_P1/index.html#link_3) who done some tests you will get it to work if you add a bigger capacitor.

psvanstrom, 2022-01-13:
@Nosen92 Feel free to update the README.md and create a pull request with the ESP32-specific info and I'll make sure to merge it 👍

closed 2022-12-19 by psvanstrom