psvanstrom/esphome-p1reader #36: Iskra AM550 - Data not published, CRC error

johansvenson, 2022-01-30:
Hi is there anyway of troubleshooting issues with the crc checksum - looking at the log the received data indeed seems off however it doesn't seem quite like just poor wiring given how the positions where we see invalid data seems to be following a pattern... well at least a bit. 

The meter is an Iskra AM550. I tried tweaking the delay and buffer size but no effect... 

[09:41:26][D][data:305]: /ISk5\2M550T-2006
[09:41:26][D][data:305]:
[09:41:26][D][data:305]: 1-0:0.9.1(094127)
[09:41:26][D][data:305]: 1-0:0.9.2(220130)
[09:41:26][D][data:305]: 0-0:96.1.0(77516889)
[09:41:26][D][data:305]: 0-2:25.6.0(ER23)
[09:41:26][D][data:305]: 0-0:25.1.0(ER23)
[09:41:26][D][data:305]: 1-0:1.8.0(027257.748*kWh)
[09:41:26][D][data:305]: 1-0:2.8.0(000000.000*kWh)
[09:41:26][D][data:305]: 1-0:3.8.0(004565.274*kvarh)
[09:41:26][D][data:305]: *kW)
[09:41:26][D][data:305]: 1-0:41.7.0(01.422*kW)
[09:41:26][D][data:305]: 1-0:42.7.0(00.000*kW)
[09:41:26][D][data:305]: 1-0:61.7.0(00.842*kW)
[09:41:26][D][data:305]: 1-0:62.7.0(00.000*kW)
[09:41:26][D][data:305]: 1-0:31.7.0(011*A)
[09:41:26][D][data:305]: 1-0:32.7.0(228.9*V)
[09:41:26][D][data:305]: 1-0:51.7.0(006*A)
[09:41:26][D][data:305]: 1-0:52.7.0(231.3*V)
[09:41:26][D][data:305]: 1-0:71.7.0(003*A)
[09:41:26][D][data:305]: 1-0:72.7.0(234.0*V)
[09:41:26][D][data:305]: 1-0:0.2.0((ER11))
[09:41:26][D][data:305]: 1-0:0.2.8((ER11))
[09:41:26][D][data:305]: !F8D6
[09:41:26][I][crc:317]: Telegram read. CRC: BF63 = F8D6. PASS = NO
[09:41:36][D][data:305]: /ISk5\2M550T-2006
[09:41:36][D][data:305]:
[09:41:36][D][data:305]: 1-0:0.9.1(094137)
[09:41:36][D][data:305]: 1-0:0.9.2(220130)
[09:41:36][D][data:305]: 0-0:96.1.0(77516889)
[09:41:36][D][data:305]: 0-2:25.6.0(ER23)
[09:41:36][D][data:305]: 0-0:25.1.0(ER23)
[09:41:36][D][data:305]: 1-0:1.8.0(027257.762*kWh)
[09:41:36][D][data:305]: 1-0:2.8.0(000000.000*kWh)
[09:41:36][D][data:305]: 1-0:3.8.0(004565.277*kvarh)
[09:41:36][D][data:305]: 1-0:4.8.0(001459.254*kvarh)
[09:41:36][D][data:305]: 000*kW)
[09:41:36][D][data:305]: 1-0:41.7.0(01.430*kW)
[09:41:36][D][data:305]: 1-0:42.7.0(00.000*kW)
[09:41:36][D][data:305]: 1-0:61.7.0(01.576*kW)
[09:41:36][D][data:305]: 1-0:62.7.0(00.000*kW)
[09:41:36][D][data:305]: 1-0:31.7.0(015*A)
[09:41:36][D][data:305]: 1-0:32.7.0(228.1*V)
[09:41:36][D][data:305]: 1-0:51.7.0(006*A)
[09:41:36][D][data:305]: 1-0:52.7.0(231.3*V)
[09:41:36][D][data:305]: 1-0:71.7.0(006*A)
[09:41:36][D][data:305]: 1-0:72.7.0(233.4*V)
[09:41:36][D][data:305]: 1-0:0.2.0((ER11))
[09:41:36][D][data:305]: 1-0:0.2.8((ER11))
[09:41:36][D][data:305]: !F1BC
[09:41:36][I][crc:317]: Telegram read. CRC: D65B = F1BC. PASS = NO
[09:41:46][D][data:305]: /ISk5\2M550T-2006
[09:41:46][D][data:305]:
[09:41:46][D][data:305]: 1-0:0.9.1(094147)
[09:41:46][D][data:305]: 1-0:0.9.2(220130)
[09:41:46][D][data:305]: 0-0:96.1.0(77516889)
[09:41:46][D][data:305]: 0-2:25.6.0(ER23)
[09:41:46][D][data:305]: 0-0:25.1.0(ER23)
[09:41:46][D][data:305]: 1-0:1.8.0(027257.780*kWh)
[09:41:46][D][data:305]: 1-0:2.8.0(000000.000*kWh)
[09:41:46][D][data:305]: 1-0:3.8.0(004565.280*kvarh)
[09:41:46][D][data:305]: 1-0:4.8.0(001459.254*kvarh)
[09:41:46][D][data:305]: 1-0:1.7.0(06.368*kW)
[09:41:46][D][data:305]: )
[09:41:46][D][data:305]: 1-0:41.7.0(01.442*kW)
[09:41:46][D][data:305]: 1-0:42.7.0(00.000*kW)
[09:41:46][D][data:305]: 1-0:61.7.0(01.580*kW)
[09:41:46][D][data:305]: 1-0:62.7.0(00.000*kW)
[09:41:46][D][data:305]: 1-0:31.7.0(015*A)
[09:41:46][D][data:305]: 1-0:32.7.0(228.2*V)
[09:41:46][D][data:305]: 1-0:51.7.0(006*A)
[09:41:46][D][data:305]: 1-0:52.7.0(231.3*V)
[09:41:46][D][data:305]: 1-0:71.7.0(006*A)
[09:41:46][D][data:305]: 1-0:72.7.0(233.5*V)
[09:41:46][D][data:305]: 1-0:0.2.0((ER11))
[09:41:46][D][data:305]: 1-0:0.2.8((ER11))
[09:41:46][D][data:305]: !8190
[09:41:46][I][crc:317]: Telegram read. CRC: 630E = 8190. PASS = NO
[09:41:56][D][data:305]: /ISk5\2M550T-2006
[09:41:56][D][data:305]:
[09:41:56][D][data:305]: 1-0:0.9.1(094157)
[09:41:56][D][data:305]: 1-0:0.9.2(220130)
[09:41:56][D][data:305]: 0-0:96.1.0(77516889)
[09:41:56][D][data:305]: 0-2:25.6.0(ER23)
[09:41:56][D][data:305]: 0-0:25.1.0(ER23)
[09:41:56][D][data:305]: 1-0:1.8.0(027257.797*kWh)
[09:41:56][D][data:305]: 0:41.7.0(01.446*kW)
[09:41:56][D][data:305]: 1-0:42.7.0(00.000*kW)
[09:41:56][D][data:305]: 1-0:61.7.0(01.581*kW)
[09:41:56][D][data:305]: 1-0:62.7.0(00.000*kW)
[09:41:56][D][data:305]: 1-0:31.7.0(015*A)
[09:41:56][D][data:305]: 1-0:32.7.0(228.3*V)
[09:41:56][D][data:305]: 1-0:51.7.0(006*A)
[09:41:56][D][data:305]: 1-0:52.7.0(231.2*V)
[09:41:56][D][data:305]: 1-0:71.7.0(006*A)
[09:41:56][D][data:305]: 1-0:72.7.0(233.4*V)
[09:41:56][D][data:305]: 1-0:0.2.0((ER11))
[09:41:56][D][data:305]: 1-0:0.2.8((ER11))
[09:41:56][D][data:305]: !24AB

johansvenson, 2022-01-30:
In the meantime i have changed to that i ignore the crc check and instead rely on the atio to returl non 0 and sending the 
i.e.:

```

`    if (atof(value) != 0)
    {
      if (strncmp(obisCode, "1.8.0", 5) == 0)
      {
        parsed->cumulativeActiveImport = atof(value);
        cumulativeActiveImport->publish_state(parsed->cumulativeActiveImport);
      }`
```

CarbonCollins, 2022-09-10:
I am also getting this issue with the same model of power meter. similar log output too:
`[21:38:04][D][data:265]: /ISk5\2M550T-2006
[21:38:04][D][data:265]: 
[21:38:04][D][data:265]: 1-0:0.9.1(203805)
[21:38:04][D][data:265]: 1-0:0.9.2(220910)
[21:38:04][D][data:265]: 0-0:96.1.0(77516706)
[21:38:04][D][data:265]: 0-2:25.6.0(ER23)
[21:38:04][D][data:265]: 0-0:25.1.0(ER23)
[21:38:04][D][data:265]: 1-0:1.8.0(008472.260*kWh)
[21:38:04][D][data:265]: 1-0:2.8.0(000000.000*kWh)
[21:38:04][D][data:265]: 1-0:3.8.0(000022.619*kvarh)
[21:38:04][D][data:265]: 1-0:4.8.0(000701.265*kvarh)
[21:38:04][D][data:265]: 1-0:1.7.0(00.500*kW)
[21:38:04][D][data:265]: 1-0:2.7.0(00.000*kW)
[21:38:04][D][data:265]: *kW)
[21:38:04][D][data:265]: 1-0:22.7.0(00.000*kW)
[21:38:04][D][data:265]: 1-0:41.7.0(00.494*kW)
[21:38:04][D][data:265]: 1-0:42.7.0(00.000*kW)
[21:38:04][D][data:265]: 1-0:61.7.0(00.013*kW)
[21:38:04][D][data:265]: 1-0:62.7.0(00.000*kW)
[21:38:04][D][data:265]: 1-0:31.7.0(000*A)
[21:38:04][D][data:265]: 1-0:32.7.0(237.7*V)
[21:38:04][D][data:265]: 1-0:51.7.0(002*A)
[21:38:04][D][data:265]: 1-0:52.7.0(241.7*V)
[21:38:04][D][data:265]: 1-0:71.7.0(000*A)
[21:38:04][D][data:265]: 1-0:72.7.0(244.0*V)
[21:38:04][D][data:265]: 1-0:0.2.0((ER11))
[21:38:04][D][data:265]: 1-0:0.2.8((ER11))
[21:38:04][D][data:265]: !1A56
[21:38:04][I][crc:276]: Telegram read. CRC: 7374 = 1A56. PASS = NO`

CarbonCollins, 2022-09-11:
I spent a bit more time looking into the log output and noticed that in both mine and @johansvenson log outputs, there seems to be an inconsistency with some of the output. for instance:

This telegram has two blank lines: 

[11:44:44][D][data:265]: /ISk5\2M550T-2006
[11:44:44][D][data:265]: 
[11:44:44][D][data:265]: 1-0:0.9.1(104445)
[11:44:44][D][data:265]: 1-0:0.9.2(220911)
[11:44:44][D][data:265]: 0-0:96.1.0(77516706)
[11:44:44][D][data:265]: 0-2:25.6.0(ER23)
[11:44:44][D][data:265]: 0-0:25.1.0(ER23)
[11:44:44][D][data:265]: 
[11:44:44][D][data:265]: 1-0:4.8.0(000704.980*kvarh)
[11:44:44][D][data:265]: 1-0:1.7.0(00.199*kW)
[11:44:44][D][data:265]: 1-0:2.7.0(00.000*kW)
[11:44:44][D][data:265]: 1-0:3.7.0(00.000*kvar)
[11:44:44][D][data:265]: 1-0:4.7.0(00.276*kvar)
[11:44:44][D][data:265]: 1-0:21.7.0(00.004*kW)
[11:44:44][D][data:265]: 1-0:22.7.0(00.000*kW)
[11:44:44][D][data:265]: 1-0:41.7.0(00.189*kW)
[11:44:44][D][data:265]: 1-0:42.7.0(00.000*kW)
[11:44:44][D][data:265]: 1-0:61.7.0(00.005*kW)
[11:44:44][D][data:265]: 1-0:62.7.0(00.000*kW)
[11:44:44][D][data:265]: 1-0:31.7.0(000*A)
[11:44:44][D][data:265]: 1-0:32.7.0(230.4*V)
[11:44:44][D][data:265]: 1-0:51.7.0(001*A)
[11:44:44][D][data:265]: 1-0:52.7.0(245.1*V)
[11:44:44][D][data:265]: 1-0:71.7.0(000*A)
[11:44:44][D][data:265]: 1-0:72.7.0(244.6*V)
[11:44:44][D][data:265]: 1-0:0.2.0((ER11))
[11:44:44][D][data:265]: 1-0:0.2.8((ER11))
[11:44:44][D][data:265]: !B587
[11:44:44][I][crc:276]: Telegram read. CRC: C779 = B587. PASS = NO

whereas this one seems to have some malformed data part way through the telegram...:

[11:45:24][D][data:265]: /ISk5\2M550T-2006
[11:45:24][D][data:265]: 
[11:45:24][D][data:265]: 1-0:0.9.1(104525)
[11:45:24][D][data:265]: 1-0:0.9.2(220911)
[11:45:24][D][data:265]: 0-0:96.1.0(77516706)
[11:45:24][D][data:265]: 0-2:25.6.0(ER23)
[11:45:24][D][data:265]: 0-0:25.1.0(ER23)
[11:45:24][D][data:265]: 1-0:1.8.0(008478.205*kWh)
[11:45:24][D][data:265]: 1-0:2.8.0(000000.000*kWh)
[11:45:24][D][data:265]: 1-0:3.8.0(000022.619*kvarh)
[11:45:24][D][data:265]: 00*kW)
[11:45:24][D][data:265]: 0(00.000*kvar)
[11:45:24][D][data:265]: 1-0:4.7.0(00.275*kvar)
[11:45:24][D][data:265]: 1-0:21.7.0(00.004*kW)
[11:45:24][D][data:265]: 1-0:22.7.0(00.000*kW)
[11:45:24][D][data:265]: 1-0:41.7.0(00.188*kW)
[11:45:24][D][data:265]: 1-0:42.7.0(00.000*kW)
[11:45:24][D][data:265]: 1-0:61.7.0(00.005*kW)
[11:45:24][D][data:265]: 1-0:62.7.0(00.000*kW)
[11:45:24][D][data:265]: 1-0:31.7.0(000*A)
[11:45:24][D][data:265]: 1-0:32.7.0(233.6*V)
[11:45:24][D][data:265]: 1-0:51.7.0(001*A)
[11:45:24][D][data:265]: 1-0:52.7.0(244.1*V)
[11:45:24][D][data:265]: 1-0:71.7.0(000*A)
[11:45:24][D][data:265]: 1-0:72.7.0(243.8*V)
[11:45:24][D][data:265]: 1-0:0.2.0((ER11))
[11:45:24][D][data:265]: 1-0:0.2.8((ER11))
[11:45:24][D][data:265]: !0904
[11:45:24][I][crc:276]: Telegram read. CRC: 3AB6 = 0904. PASS = NO

you can also see this in the telegram I posted above too...

CarbonCollins, 2022-11-17:
Hi again,

I've been looking into the issue a bit more as of late and think I may have come to a idea as to what was occuring.

It seems that the AM550 was pushing data faster than what the reader was processing causing the internal serial buffer to fill up and drop data. This is mainly from the way the data was malformed in the previous telegrams I posted. I had success doing one of two things:

1. decrease the delay at the end of the while loop from 40 to 10 which makes the reader empty the serial buffer more frequently.

2. Increase the internal serial buffer from the 256 default to 512/1024 by adding `Serial.setRxBufferSize(512);` in the setup function.

In either of these two cases I did not get CRC check errors nor any malformed telegrams. Ive been running it live for the last month or so without any issues here!

I'm not sure which option would be the better choice to raise as a PR though, any thoughts on this?