psvanstrom/esphome-p1reader #43: Debug not showing expected output.

MissMusic, 2022-04-01:
So i just installed this on my new meter, with a D1 mini.

This is the output I'm getting. Any idéa on what could be the issue?

With pin not inverted:

```
[14:25:21][D][data:265]: 
[14:25:21][D][data:265]: 
[14:25:21][D][data:265]: 
[14:25:30][D][data:265]: ~\xa2CA\x83\x85\xeb\xe6\xe7
[14:25:30][D][data:265]: 
[14:25:30][D][data:265]: 
[14:25:30][D][data:265]: \xff!	
[14:25:30][D][data:265]: #	
[14:25:30][D][data:265]: 
\xee
[14:25:30][D][data:265]: 
[14:25:30][D][data:265]: 
[14:25:31][D][data:265]: 
```

And with pin invted:
```
[14:28:20][D][data:265]: \xa0\x97\xf5\xbd\xf9v=
[14:28:20][D][data:265]: \x86\xf8\xe1\xff\xff\xff\xff\xfd\xc9\xfb\xfb\xed\xf3\xff\xff\xfd\xff\xff{^3_\xf5\xe5\xc7\xd7\xfe\xbf\xed\xf3\xfd\xff\xfd\xf1\xff\xbe\xff\xff\xff\xfb\xfb\xe1\xff\xd3\xc9\xfb\xf9\xed\xf3\xfd\xff\xfb\xf1\xff

[14:28:20][D][data:265]: \xbe\xffӻ\xbf\xbf\xff\xd3\xc9\xfb\xf9\xed\xf3\xfd\xff\xf9\xf1\xff\xbe\xff\xff\xff\xfb\xfb\xe1\xff\xd3\xc5\xfb\xf9\xed\xf3\xfd\xff\xf7\xf1\xff\xbe\xff\xeb\xb9\xfb\xfb\xe1\xff\xd3\xc5\xfb\xf9\xed\xf3\xfd\xff\xc1\xf1\xff

[14:28:20][D][data:265]: \xcc\xfb\xe1\xba\xdb\xf9\xed\xf3\xfd\xff\x99\xf1\xff\x94\xbf\xbf\xba\xdb\xf9\xed\xf3\xfd\xffq\xf1\xffe\xfb\xe1\xba\xdb\xf9\xed\xf3\xfd\xff\xbf\xf1\xff\xbb\xef\xb6\xfb\xe1\xba\xdb\xf9\xed\xf3\xfd\xff

[14:28:20][D][data:265]: \x97\xf1\xff\xbb/\xfb\xfb\xe1\xba\xdb\xf9\xed\xf3\xfd\xffo\xf1\xff\xbb/\xb6\xfb\xe1\xba\xdb\xf9\xed\xf3\xfd\xff\xd5\xf1\xff\xbe\xff\xff\xff\xfb\xfb\xe1\xff\xd3\xc9\xfb\xf9\xed\xf3\xfd\xff\xd3\xf1\xff\xbe\xff

[14:28:20][D][data:265]: ﭿ\xbf\xff\xd3\xc9\xfb\xf9\xed\xf3\xfd\xff\xd1\xf1\xff\xbe\xff\xff\xff\xfb\xfb\xe1\xff\xd3\xc5\xfb\xf9\xed\xf3\xfd\xff\xcf\xf1\xff\xbe\xff\xf9\xa3\xfb\xfb\xe1\xff\xd3\xc5\xfb\xf9\xed\xf3\xfd\xff\xad\xf1\xff\xbe\xff\xff

[14:28:20][D][data:265]: \xff\xfb\xfb\xe1\xff\xd3\xc9\xfb\xf9\xed\xf3\xfd\xff\xab\xf1\xff\xbe\xff\xf3\xfb\xfb\xe1\xff\xd3\xc9\xfb\xf9\xed\xf3\xfd\xff\xa9\xf1\xff\xbe\xff\xff\xff\xfb\xfb\xe1\xff\xd3\xc5\xfb\xf9\xed\xf3\xfd\xff\xa7\xf1\xff\xbe\xff\xf9

[14:28:20][D][data:265]: g\xbf\xbf\xff\xd3\xc5\xfb\xf9\xed\xf3\xfd\xff\x85\xf1\xff\xbe\xff\xff\xff\xfb\xfb\xe1\xff\xd3\xc9\xfb\xf9\xed\xf3\xfd\xff\x83\xf1\xff\xbe\xff\xf1鿿\xff\xd3\xc9\xfb\xf9\xed\xf3\xfd\xff\x81\xf1\xff\xbe\xff\xff\xff\xfb

Z\xbf\xbf

[14:28:21][D][data:265]: \xff\xd3\xc3\xfb\xf9\xed\xf3\xfd\xff\xf9\xef\xff\xbe\xff\xff\xff\xfb\xfb\xe1\xffӿ\xfb\xf9\xed\xf3\xfd\xff\xf7\xef\xff\xbe\xf1uj\xbf\xbf\xffӿ\xe1+
```

Xoliu, 2022-04-25:
Did you solve it? I get exactly the same data as you, when the pin is inverted.

MissMusic, 2022-04-26:
Yes and no.

I figured out that the issue is that the Aidon meter uses a different communication standard. To rule out my own hardware i bought a SlimmeLezer+ but still no go.

I also bought a P1IB, that had a setting specific for the Aidon meter, and that worked!

After setting up the P1IB i was also shared this library that are supposed to get the SlimmeLezer to work with the Aidon meter, i have not tried it, but i expect that it could work also for this P1 reader:
https://mega.nz/file/U4UTULpJ#KhJYfEhRiUQ8CI5l7ypDbKKFp0NIb4si4qWBB8GGI6I

Good luck!

Xoliu, 2022-04-26:
Thanks for the reply! I have a hard time getting hardware delivered to me at the time. I would love to buy a finished product to just plug in but I will try out the code you provided for now and see if it makes any difference. I can post an update here if you are interested in knowing how it went. Take care.

Xoliu, 2022-04-26:
Wow, It actually worked. That code and a small change to invert the pin and voilá.
I also added a small capacitor between +5V and Ground since I saw it flicker.

[image]