psvanstrom/esphome-p1reader #44: Changes with Slimmelezer and Aidon 6534

MrGlad8, 2022-04-15:
First of all, thank a lot for this code. Helped me a lot!
I bought Slimmelezer Plus for my Aidon 6534 just to realise afterwards that the energy meter doesn't follow the standard protocol.
But with your p1reader-code with some modifications I got it to to work. Thanks to Skracke on byggahus forum and jsuanet at HA forums. Let me show you how I got it to work!
The changes I did in order to get the Slimmelezer Plus to work with my Aidon 6534 meter was:

### **In file "p1reader.h" =**

- #define BUF_SIZE 50 to #define BUF_SIZE 60
- 
- In every place where "readBytes" occurred I changed to "Serial.readBytes". Was like in 5-6 places, I used the search for "bytes".
- 
- Also in some placed there was "readBytesUntil" and that I changed to "Serial.readBytesUntil".

### **And in file "p1reader.yaml" I changed:**

- tx_pin: TX to tx_pin: D8
- rx_pin: RX to tx_pin: D7
- 
- 
- I also changed the name from "electricity_meter" to "slimmelezer". Not sure why but jsuanet on the forums said OTA update would fail otherwise. I did the change and have not had any problems yet.
- 
- Also changed board from "nodemcu" to "d1_mini"
- 
- 
- I also added:
- 
- web_server:
-   port: 80
- 
- In order to be able to surf into the IP number of my Slimmelezer.
- 
- 
- And lastly I added to the sensor "Cumulative Active Import"
- state_class: total_increasing
- device_class: energy
- 
- So that it could be used as an energy-meter in HA.

I hope this can help someone else in my position with a Slimmelezer Plus and a Aidon 6534 energy meter that doesnt follow standard protocol :)

denmol, 2023-04-12:
Hi 

I am having truble geting data fram my skekraft labeled aidon meter this is the data i get: 

12:59:20 | [D] | [data:265] | 
-- | -- | -- | --
12:59:20 | [D] | [data:265] | �!	
12:59:20 | [D] | [data:265] | #	
12:59:20 | [D] | [data:265] | 
12:59:20 | [D] | [data:265] |  
12:59:20 | [D] | [data:265] | 
12:59:20 | [D] | [data:265] | 
12:59:21 | [D] | [data:265] | 
12:59:29 | [D] | [data:265] | ~�CA�����
12:59:30 | [D] | [data:265] | 
12:59:30 | [D] | [data:265] | �
12:59:30 | [D] | [data:265] | �!	
12:59:30 | [D] | [data:265] | #	
12:59:30 | [D] | [data:265] | 
12:59:30 | [D] | [data:265] |  
12:59:30 | [D] | [data:265] | 
12:59:30 | [D] | [data:265] | 
12:59:31 | [D] | [data:265] | 
12:59:36 | [D] | [sensor:126] | 'SlimmeLezer Uptime': Sending state 404.45999 s with 0 decimals of accuracy
12:59:39 | [D] | [data:265] | ~�CA�����
12:59:40 | [D] | [data:265] | 
12:59:40 | [D] | [data:265] | 
12:59:40 | [D] | [data:265] | �!	
12:59:40 | [D] | [data:265] | #	
12:59:40 | [D] | [data:265] | 
12:59:40 | [D] | [data:265] |  
12:59:40 | [D] | [data:265] | 
12:59:40 | [D] | [data:265] | 
12:59:41 | [D] | [data:265] | 
12:59:49 | [D] | [data:265] | ~�CA�����
12:59:50 | [D] | [data:265] | 
12:59:50 | [D] | [data:265] | �
12:59:50 | [D] | [data:265] | �!	
12:59:50 | [D] | [data:265] | #	
12:59:50 | [D] | [data:265] | 
12:59:50 | [D] | [data:265] |  
12:59:50 | [D] | [data:265] | 
12:59:50 | [D] | [data:265] | 
12:59:51 | [D] | [data:265] | 
12:59:52 | [D] | [sensor:126] | 'SlimmeLezer Wi-Fi Signal': Sending state -73.00000 dBm with 0 decimals of accuracy
12:59:59 | [D] | [data:265] | ~�CA�����
13:00:00 | [D] | [data:265] | 
13:00:00 | [D] | [data:265] | �
13:00:00 | [D] | [data:265] | �!	
13:00:00 | [D] | [data:265] | #	
13:00:00 | [D] | [data:265] | 
13:00:00 | [D] | [data:265] |  
13:00:00 | [D] | [data:265] | 
13:00:00 | [D] | [data:265] | 
13:00:01 | [D] | [data:265] | 

I have found some information about the meter but my coding skills are none so a would appreciate some help.

Interface desription from skekraft: https://www.skekraft.se/wp-content/uploads/2021/03/Aidon_Feature_description_RJ45_HAN_Interface_EN.pdf

and some example to read the data 
https://github.com/OlleRatu/han-rj45/blob/main/HAN_aidon_rj45_example/han.py

Megamannen, 2023-04-19:
> Hi
> 
> I am having truble geting data fram my skekraft labeled aidon meter this is the data i get .....

Try this: https://github.com/psvanstrom/esphome-p1reader/pull/70