psvanstrom/esphome-p1reader #47: No data from Kamstrup Omnipower

tuomaz, 2022-05-21:
Hi!
I'm using a Slimelezer+ configured as https://github.com/psvanstrom/esphome-p1reader/issues/45 together with a Kamstrup Omnipower. The Kamstrup Omnipower has new updated firmware and the setting "HAN push list" set to "HAN P1 (värde 2)".

I get every 10 second but only this:

```
[10:04:56][D][data:265]: /KAM5
[10:04:57][D][data:265]:
[10:05:06][D][data:265]: /KAM5
[10:05:07][D][data:265]:
[10:05:16][D][data:265]: /KAM5
[10:05:17][D][data:265]:
```

Is it something I try do change / debug this? Or can it be a bug in the firmware of the Kamstrup Omnipower?

fredrik-tuomas-lv, 2022-06-22:
Solved by modding my cable and enabling +5V on PIN1.

MattiasPie, 2022-07-16:
@fredrik-tuomas-lv Hi Fredrik, I'm having the same setup as you and have configured my Slimelezer as #45. However I'm getting nothing not even power from the Rj12 cable. I meassured the Pins and could detect 5v on some of them.
Can you please help me how to trouble shoot this?
I will also try to reach you by email on your first account and I hope you don't mind.
Many thanks in advance.
/Mattias

runevad, 2022-07-16:
> @fredrik-tuomas-lv Hi Fredrik, I'm having the same setup as you and have configured my Slimelezer as #45. However I'm getting nothing not even power from the Rj12 cable. I meassured the Pins and [...]

Unfortunately you need to supply the 5V from an external source to get the meter to start sending data. The Kamstrup meters give 5V when they have the HAN RJ45 module but unfortunately not on the P1 module, no idea why...

tuomaz, 2022-07-16:
Yes, you need to supply +5V externally to get it to work. See here also: https://www.byggahus.se/forum/threads/elmaetare-h1-port-p1-port-han-port-manga-namn-verkar-florera.384608/page-52#post-4781520

MattiasPie, 2022-07-18:
Ouch… my meter is outside with no power socket nearby. Maybe it would be easier to switch to this https://www.utomhusliv.se/produkt/kamstrup-han-modul/ to be able to get power?
Do you guys have any coments on that approach?

lorrob, 2022-09-02:
On my Kamstrup meter I needed to splice 5V on pin 1 and ground on pin 6 to wake up the module. Then to get data i needed to set the baud rate to 2400. Now i get the data but need to fix the parsing.

MattiasPie, 2022-09-03:
I just realized pasted wrong link. This:
https://amsleser.no/hjem/1-pow-k-kamstrup.html

I ended up buying this one and I would highly recomend this instead of P1Han module and a slimmelezer. It’s even cheaper and have integration to Entso-E for day ahead prices. Mqtt with home Assistant auto discover etc.

ekamagn, 2022-10-11:
Drog min rj12 kabel genom fönstret så elektronik hamnar inomhus och där
brukar det finnas vägguttag i närheten.
Kanske ett alternativ för att fixa 5v?
Mvh Magnus

runevad, 2023-05-10:
> 

Yes, this product work fine for all Kamstrup Omnipower **BUT only with the original NVE firmware** that sends data as an AMS meter, hence via M-BUS and with the Norwegian standard HAN protocol. 

Nowadays Kamstrup (at least with my grid operator) can supply the Kamstrup Omnipower meter with a firmware update so that the meter can be set to **either HAN or P1 depending on what the customer wants**. It's just a setting in the Omnia software that the grid operator has. My own Kamstrup Omnipower now sends data (with the P1 port) and now works with the [SlimmeLezer](https://www.zuidwijk.com/product/slimmelezer/) as any other P1 meter.

tejasitraj, 2023-06-02:
I am supplying my slimmelezer with usb-c power, so when i measure the voltage across pin 1 and pin 6 of the rj12, i get 4,8V. Do I still need to splice 5V into the RJ12 cable? 
My electricity company has confirmed that the port is active as per HAN-P1 standard.
Currently I see the following output:
[image]

tuomaz, 2023-06-02:
@tejasitraj 

If I remember correctly that output is similar to what I got before I fixed my cable. Every 10 second you should get data (import, export, voltage etc)

tejasitraj, 2023-06-02:
@tuomaz before you fixed your cable, did you power your slimmelezer+ with usb-c in its port? So now that you have fixed your cable, do you power the slimmelezer+ in its own port plus 5V on the RJ12 cable? Any chance you can show a picture?

tejasitraj, 2023-06-03:
Ok so I've solved it too. Like you say, the HAN P1 module requires power too, it is not enough to only feed the slimmelezer with usb-c. Like you said, I spliced into my RJ12 cable and attached 2 cables, to pin 1 and 6 respectively. And fed those with 5V external power. Then the meter woke up and started transmitting data.

nilsd, 2023-11-13:
Hello! I have an Omnipower as well. With 9600 baud rate I managed to get this far, without any 5v modification:

[image]

0 kw is not correct though, does this screenshot tell anyone anything useful? I've confirmed with my electric company that they are sending unencrypted data. Given this screenshot, could I still be having issues with unsufficient 5V power or have I passed the point where it's probably the meter that hasn't started sending data?

I'm running a D1 Mini on a Weigu SmartyReader 2.2 board.

Thank you!