psvanstrom/esphome-p1reader #49: No data from S34U18 on Slimmelezer+

pretori, 2022-06-15:
Hi

I have flashed my unit (Slimmelezer+) according to https://github.com/psvanstrom/esphome-p1reader/issues/45 .
When connecting it to my Vattenfall meter S34U18 I get no data at all.

Is it possible to debug the unit? And se what's happening?

pretori, 2022-06-16:
I have measured the pin voltage om the RJ12, 
Pin 1 VCC to pin 6 GND = 5V, 
Pin 2 DATA REQUEST to pin 6 GND = 5V
So far so good :-)

Pin 5 DATA Out to pin 3 DATA GND = constant 2549mV
That look crazy.....

dala318, 2022-06-16:
Is the port activated?
On my E.On device I had to call their support and have them activate it remotely.

pretori, 2022-06-16:
I have asked Vattenfall 2 times and they say that it’s open. But I’m still not sure🤔

djmGH01, 2022-08-10:
> 

I do have the same issue.
To me it looks that rx is shared with the USB communication component and therefor cannot pull the signal "low".
Didi you already have any solution to this?

pretori, 2022-08-10:
My slimmelezer is resting in a drawer, I got another p1 reader that worked out of the box

stewed, 2022-09-05:
Which one did you get? I have a Vattenfall S34U18 too.

Gronsak, 2022-12-20:
Running into this problem as well, though mine is branded with a subsidiary of Vattenfall.
Tried adding debug to the UART setting to see if any data is being sent at all but so far nothing
```
uart:
  id: uart_bus
  rx_pin: D7
  baud_rate: 115200
  debug:
    direction: RX
    after:
      bytes: 150
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
```
Has anyone managed to figure anything about this or has a better solution on how to debug the issue?

djmGH01, 2022-12-21:
Do you use a wemos d1?Probably D7 is then not supporting the high Baud rate. Try to use RxD (check if usb pull up should be removed!)

Gronsak, 2022-12-21:
Using the Slimmelezer+ just as in the original issue, configured as #45 suggests. 
Would be weird if it didn't support the baudrate since DSMR 5 that it is designed to be used on is a derivative of IEC 62056-21 Mode D and, if I understand it correctly, is running the same baud rate. At least from what I can see in the [original configuration](https://www.zuidwijk.com/slimmelezer-configuration/) for the Slimmelezer+ it is running at the same rate.
```
...

uart:
  baud_rate: 115200
  rx_pin: D7

...
```
Slimmelezer+ is running on a ESP-12F and D7 which in the spec sheet is listed with UART0_CTS functionality and I can't find any reference to a limitation on the baud rate on that. But I am a novice.

I'll look into the USB pull up but would be nice to know if anyone has hade any success with the Slimmelezer+ before I start de-soldering stuff (again, I'm no more than a novice when it comes to hardware)

Gronsak, 2023-01-03:
Figured it out, the problem was on the energy company side. 
For some reason the port activation hadn't gone through so there was nothing wrong with the hardware or configuration.
Working like a charm now.