psvanstrom/esphome-p1reader #55: P1 Splitter for two or more devices

bortek, 2022-10-04:
Hi

Has anyone tried to use P1 splitter or connect two devices to a single P1 port? I need to connect two devices to my meter. One devices is obviously this esphome reader and another one is a load balancer for EV car charging called "Easee Equalizer". 

I have found these P1 splitters which claim to work . Anyone has any experience or know if they will or will not work? 

https://www.homewizard.com/shop/active-p1-splitter/
https://webshop.cedel.nl/P1-Kabel-Splitter#
https://www.bol.com/nl/nl/p/p1-splitter-voor-laadpalen/9300000021044239/#modal_open

Would rather build myself one but cannot find any working DIY solution.

grodansparadis, 2022-10-04:
You can take out a total of 250mA  but apart from that any standard low cost splitter passive should do IMHO.

tedenda, 2022-10-04:
I found this device a while ago:
[Active P1 splitter](https://www.homewizard.com/shop/active-p1-splitter/)
I have not tested it so it's just a link to another product.

bortek, 2022-10-04:
Yes this one was first among my links.

Now i just a reply in another forum that Easee Equalizer supports home assistans and can feed all the data that esphome doungle does. So if I use Equalizer then I won't need to use sny splitter.

closed 2022-10-04 by bortek

AndersHoglund, 2023-01-06:
Had the same problem with Easee Equalizer, modified my Wemos D1-mini not too load signal line. Original design limits Vih to the base-emitter voltage, 0,7V. Too low for Equalizer. 
Remove pull-up resistor, add series resistor on data line.
[image]