psvanstrom/esphome-p1reader #56: No signal D1 mini and Landis+Gyr E360

GhostNr1, 2022-10-04:
I was reading thru the #4 and they was talking about using a 1K resistor. Did you change anything from the schamatics that is shown here or is it the same. When I'm looking at the signal it looks like I only get around 1V signal from the data pin so was wounding if I need to lowering the resistor from 5V from 4.7K to 1K.

closed 2022-10-11 by GhostNr1

ulfwin, 2022-10-13:
Hi @GhostNr1. I'm currently looking at exactly the same thing, and I'm confused... The README point to the the comment in the issue, as if an action is needed, but further down in the issue it says the change is merged. 

It seems bortek modified the hardware to control the RTS pin, but I'm not sure that's needed anymore.

Did you figure this out already?

ulfwin, 2022-10-21:
To answer my own question:
I did no special hardware or software change, and it's working fine for several days now. I do see missing samples some times, which might suggest CRC errors (or perhaps bad wifi connection), but no complete breakdown as described in the issue.

I could of course also be that the issue was fixed in my electricity meter...

GhostNr1, 2022-10-24:
@ulfwin I'm to emberrased to tell you my problem. I had turn the transistor in the wrong direction after looked at it many times to get it right. I also had set it to invert the pins so I get only a lot of crap in the beginning. But when I sorted that out everything worked great. Have had it without any problem for some days now. Glad it worked out for you too.

ulfwin, 2022-10-24:
😆 We've all been there. Thanks for the update!