psvanstrom/esphome-p1reader #58: Hungarian Sanxing SX631 Read issue

deesnow, 2022-10-08:
I tried to use this ESPHome P1 reader on Wemos D1 mini, unfortunately it does not parse the serial communication.

This is what I got on Serial port (originally I was able to capture it with Tasmota)
`"0-0:17.0.0(90.000*kW)\r"
"1-0:1.8.0(001540.369*kWh)\r"
"1-0:1.8.1(000695.519*kWh)\r"
"1-0:1.8.2(000844.850*kWh)\r"
"1-0:1.8.3(000000.000*kWh)\r"
"1-0:1.8.4(000000.000*kWh)\r"
"1-0:2.8.0(005431.496*kWh)\r"
"1-0:2.8.1(003888.665*kWh)\r"
"1-0:2.8.2(001542.831*kWh)\r"
"1-0:2.8.3(000000.000*kWh)\r"
"1-0:2.8.4(000000.000*kWh)\r"
"1-0:3.8.0(000274.046*kvarh)\r"
"1-0:4.8.0(000732.779*kvarh)\r"
"1-0:5.8.0(000154.115*kvarh)\r"
"1-0:6.8.0(000119.931*kvarh)\r"
"1-0:7.8.0(000248.954*kvarh)\r"
"1-0:8.8.0(000483.825*kvarh)\r"
"1-0:15.8.0(006971.872*kWh)\r"
"1-0:32.7.0(234.0*V)\r"
"1-0:52.7.0(232.4*V)\r"
"1-0:72.7.0(229.9*V)\r"
"1-0:31.7.0(000*A)\r"
"1-0:51.7.0(000*A)\r"
"1-0:71.7.0(000*A)\r"
"1-0:13.7.0(0.992)\r"
"1-0:33.7.0(0.781)\r"
"1-0:53.7.0(0.989)\r"
"1-0:73.7.0(0.646)\r"
"1-0:14.7.0(50.02*Hz)\r"
"1-0:1.7.0(00.000*kW)\r"
"1-0:2.7.0(00.256*kW)\r"
"1-0:5.7.0(00.000*kvar)\r"
"1-0:6.7.0(00.051*kvar)\r"
"1-0:7.7.0(00.083*kvar)\r"
"1-0:8.7.0(00.000*kvar)\r"
"0-0:98.1.0(221001000000S)(001522.200*kWh)(000694.331*kWh)(000827.869*kWh)(005400.112*kWh)(003880.377*kWh)(001519.735*kWh)(000272.325*kvarh)(000722.372*kvarh)(000153.011*kvarh)(000119.314*kvarh)(000246.253*kvarh)(000476.119*kvarh)(006922.318*kWh)(03.968*kW)(03.968*kW)(02.676*kW)(04.960*kW)(04.960*kW)(04.304*kW)\r"
")\r"
"!75B4\r"`

Because of the long string in **0-0:98.1.0**  OBIS code on Tasmota I had to apply seriall buffer to 1700. I have tried to add this to configuration but did not recevice correct information.

`uart:
  id: uart_bus
  rx_pin: RX
  rx_buffer_size: 1700`

I saw the following in the logs:

`INFO Reading configuration /config/esphome/p1-esphome.yaml...
INFO Starting log output from 192.168.0.149 using esphome API
INFO Successfully connected to 192.168.0.149
[20:11:24][I][app:102]: ESPHome version 2022.9.4 compiled on Oct  8 2022, 20:05:20
[20:11:24][C][wifi:502]: WiFi:
[20:11:24][C][wifi:360]:   Local MAC: C8:C9:A3:XX:XX:XX
[20:11:24][C][wifi:361]:   SSID: 'Jtarjanyi-HA'[redacted]
[20:11:24][C][wifi:362]:   IP Address: 192.168.x.x
[20:11:24][C][wifi:363]:   BSSID: D0:6E:DE:5A:3C:50[redacted]
[20:11:24][C][wifi:365]:   Hostname: 'p1-esphome'
[20:11:24][C][wifi:367]:   Signal strength: -77 dB ▂▄▆█
[20:11:24][C][wifi:371]:   Channel: 4
[20:11:24][C][wifi:372]:   Subnet: 255.255.255.0
[20:11:24][C][wifi:373]:   Gateway: 192.168.x.x
[20:11:24][C][wifi:374]:   DNS1: 0.0.0.0
[20:11:24][C][wifi:375]:   DNS2: 0.0.0.0
[20:11:24][C][logger:275]: Logger:
[20:11:24][C][logger:276]:   Level: DEBUG
[20:11:24][C][logger:277]:   Log Baud Rate: 0
[20:11:24][C][logger:278]:   Hardware UART: UART0
[20:11:24][C][uart.arduino_esp8266:102]: UART Bus:
[20:11:24][C][uart.arduino_esp8266:104]:   RX Pin: GPIO3
[20:11:24][C][uart.arduino_esp8266:106]:   RX Buffer Size: 1700
[20:11:24][C][uart.arduino_esp8266:108]:   Baud Rate: 115200 baud
[20:11:24][C][uart.arduino_esp8266:109]:   Data Bits: 8
[20:11:24][C][uart.arduino_esp8266:110]:   Parity: NONE
[20:11:24][C][uart.arduino_esp8266:111]:   Stop bits: 1
[20:11:24][C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[20:11:24][C][captive_portal:088]: Captive Portal:
[20:11:24][C][mdns:100]: mDNS:
[20:11:24][C][mdns:101]:   Hostname: p1-esphome
[20:11:24][C][ota:089]: Over-The-Air Updates:
[20:11:24][C][ota:090]:   Address: 192.168.x.x:8266
[20:11:24][C][ota:093]:   Using Password.
[20:11:24][C][api:138]: API Server:
[20:11:24][C][api:139]:   Address: 192.168.x.x:6053
[20:11:24][C][api:141]:   Using noise encryption: YES

[20:11:27][D][data:265]: 0-0:98.1.0(221001000000S)(001522.200*kWh)(000694.331*kWh)(00

[20:11:27][D][data:265]: 0827.869*kWh)(005400.112*kWh)(003880.377*kWh)(001519.735*kWh

[20:11:27][D][data:265]: )(000272.325*kvarh)(000722.372*kvarh)(000153.011*kvarh)(0001

[20:11:27][D][data:265]: 19.314*kvarh)(000246.253*kvarh)(000476.119*kvarh)(006922.318

[20:11:27][D][data:265]: *kWh)(03.968*kW)(03.968*kW)(02.676*kW)(04.960*kW)(04.960*kW)

[20:11:27][D][data:265]: 0-0:96.13.0(\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff

[20:11:27][D][data:265]: \xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff

[20:11:27][D][data:265]: \xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff

[20:11:27][D][data:265]: \xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff

[20:11:27][D][data:265]: \xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff

[20:11:27][D][data:265]: \xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff

[20:11:27][D][data:265]: \xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff

[20:11:27][D][data:265]: \xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff

[20:11:27][D][data:265]: \xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff

[20:11:27][D][data:265]: \xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff

[20:11:27][D][data:265]: \xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff

[20:11:27][D][data:265]: \xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff

[20:11:27][D][data:265]: \xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff

[20:11:27][D][data:265]: \xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff

[20:11:27][D][data:265]: \xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff

[20:11:27][D][data:265]: \xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff

[20:11:27][D][data:265]: \xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff

[20:11:27][I][crc:276]: Telegram read. CRC: 4528 = BF59. PASS = NO`

Any iseda what should I try ?

zsolesz79, 2022-10-29:
Having the same issue, any update on this?

deesnow, 2022-10-29:
> Having the same issue, any update on this?

I had no success with this integration.
Küldj egy e-mail privátban és megküldöm a megoldást ;)
janos.tarjanyi@gmail.com