psvanstrom/esphome-p1reader #63: D1 Mini suddenly needs external power to start

flaektrem, 2023-01-06:
Running this on D1 Mini and has always been powered only from the P1 port (Vattenfall). Everything has been working flawlessly until I updated ESP HA addon and flashed devices with latest ESP fw (2022.12.3)
After successful flash, the device did not start and after hours of investigation I realized that after update my D1 mini needs external power to "jump start". After up and running I can remove external power and it keeps alive from P1 port.

Is there any breaking change I've totally missed?

Deichscheich, 2023-03-12:
I don't have prior experience with this, just trying to connect my D1 mini to a L&G 360 for the first time today.

The blue LED on the D1 mini briefly lights up, then I hear a relais in the meter. Power comes on for a second, then off, then on again.

Did you power your device with USB, in parallel to the 5V from the P1 port?

314159-r, 2023-04-30:
@flaektrem What design of P1 interface are you using?
If your design have a large capacitor on the power supply, like Weigu's SmartyReader, with 1000µF, I believe reducing that to like 1-10 µF would be appropriate. 1000µF is way overkill here in my opinion.

The power supply in the P1 port  is voluntary and its power capacity is unspecified. 
I believe that many people who have reported problem with cold booting, suffer from a to slow too slow voltage rise when power up, due to the current limit in the power supply and a large capacitor. A smaller capacitor will make the rise time faster...

The updated ESPHome version might have changed the boot behavior of the ESP, and if your previous version barely managed to boot - it now fails... 

I'm currently running p1reader on a D1 mini with 10µF and ESPhome 2023.4.2. Stable and with no boot problem.
I have also recently tested to connect and run a XIAO ESP32C3 with only 270nF on the power supply from the P1 port. It starts and seems to run fine. But that is less than I would normally recommend. 
Unfortunately we have a compiling error with the ESP32C3 and p1reader at the moment... #72

erikarenhill, 2023-06-26:
I had problems booting up a wemos d1 mini as well, changed board to another clone from another supplier and it worked straight out of the box. Some clones seem to have low quality components