psvanstrom/esphome-p1reader #65: Extend the esp to act as active repeater to connect additional P1 device

pumpagubben, 2023-02-09:
A small addition to the code will make the p1reader to repeat the signal to the TX pin. This enables additional devices to connect and use the p1 port and no need to buy a separate active r12 splitter. 

The following is successfully implemented with a Tibber Pulse and a NodeMCU as p1reader/repeater: 

1. Set TX pin to inverted p1reader.yaml: 
```
# Set TX pin to inverted to repeat signal correctly
uart:
  id: uart_bus
  tx_pin:
    number: GPIO1
    inverted: yes
  rx_pin: RX
  baud_rate: 115200
```

2. In the p1reader.h add a line to write the received signal back to the uart_bus, add Serial.write(buffer); after the newline is put back: 
```
            // put newline back as it is required for CRC calculation
            buffer[len] = '\n';
            buffer[len + 1] = '\0';
			
            // Write back the message to TX pin as a repeater: 
            Serial.write(buffer);

            // if we've reached the CRC checksum, calculate last CRC and compare
            if (buffer[0] == '!') {
```

3. Wiring for next device: 
Connect a new r12 port or cable and connect +5V, GND, Data_GND accordingly and Data_OUT is connected to ESP TX pin. 
Leave the Data_Request unconnected (at least Tibber Pulse will not process if this is pulled high).

psvanstrom, 2023-02-18:
This is great, would you mind making a PR of these changes? 👍

robinostlund, 2023-04-05:
@pumpagubben are you able to make a pr for this? This is exactly what i am searching for as i also have a tibber pulse 😊