psvanstrom/esphome-p1reader #71: Instability Frequent Drop out

dilipva, 2023-04-20:
I have been using 8266 as P1 reader since 6 months without any issue until now. ESP crashes pretty much every 2nd readings. I replaced with new 8266 and I updated to the latest esphome since then and even moved a WIFI AP literally next to the EL box but nothing fixed the issue. I have now reduced the wifi output_power from default 20 to 17 and this reduced the number of crashes. I am just wondering what might have caused this sudden change in behavior. I also have capacitor at the power line.

Megamannen, 2023-04-20:
Try powering it separately and not with the P1 port

devilmastah, 2023-05-15:
Running into the same issue here, becomes unavailable for a small second then is back, also with external power

dilipva, 2023-05-15:
I am using https://github.com/Beaky2000/esphome-p1mini with 10s in between reads. It’s working so far without any issue.