psvanstrom/esphome-p1reader #74: Aidon meter and slimmelezer-wt32-eth01

erikfluff, 2023-05-10:
Hi! I am trying to install this code on a slimmelezer-wt32-eth01, for being able to read data from my Aidon meter using HDLC. I have modified the code and i find it on the network but i only get "[hdlc:474] Expected 12 bytes, got 0 bytes - out of sync. Returning" in the log. With the original firmware for slimmelezer-wt32-eth01 i get "[dsmr:092] | Timeout while reading data for telegram" 

The P1 port is activated, I have connected a Easee Equalizer and that gets correct data

```
esphome:
  name: esp-p1reader
  includes:
    - p1reader.h

esp32:
  board: esp32dev
  framework:
    type: arduino

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16

# Enable logging
logger:
  level: DEBUG
  baud_rate: 0 # disable logging over uart
  
# Enable Home Assistant API
api:
  encryption:
    key: !secret encryption_key
    
ota:
  password: !secret ota_password
  
web_server:
  port: 80

uart:
  id: uart_bus
  baud_rate: 115200
  rx_pin: GPIO5
 

# If your electricity meter is an Aidon, which use the older
# Branschstandard (1.2) where the data is HDLC-formatted
# Change the first line after lambda to
# auto_meter_sensor = new P1ReaderHDLC(id(uart_bus));
# else
# auto meter_sensor = new P1Reader(id(uart_bus));
sensor:
- platform: custom
  lambda: |-
    auto meter_sensor = new P1ReaderHDLC(id(uart_bus));
    App.register_component(meter_sensor);
    return {
      meter_sensor->cumulativeActiveImport,
      meter_sensor->cumulativeActiveExport,
      meter_sensor->cumulativeReactiveImport,
      meter_sensor->cumulativeReactiveExport,
      meter_sensor->momentaryActiveImport,
      meter_sensor->momentaryActiveExport,
      meter_sensor->momentaryReactiveImport,
      meter_sensor->momentaryReactiveExport,
      meter_sensor->momentaryActiveImportL1,
      meter_sensor->momentaryActiveExportL1,
      meter_sensor->momentaryActiveImportL2,
      meter_sensor->momentaryActiveExportL2,
      meter_sensor->momentaryActiveImportL3,
      meter_sensor->momentaryActiveExportL3,
      meter_sensor->momentaryReactiveImportL1,
      meter_sensor->momentaryReactiveExportL1,
      meter_sensor->momentaryReactiveImportL2,
      meter_sensor->momentaryReactiveExportL2,
      meter_sensor->momentaryReactiveImportL3,
      meter_sensor->momentaryReactiveExportL3,
      meter_sensor->voltageL1,
      meter_sensor->voltageL2,
      meter_sensor->voltageL3,
      meter_sensor->currentL1,
      meter_sensor->currentL2,
      meter_sensor->currentL3
    };
  sensors:
  - name: "Cumulative Active Import"
    unit_of_measurement: kWh
    accuracy_decimals: 3
    state_class: "total_increasing"
    device_class: "energy"
  - name: "Cumulative Active Export"
    unit_of_measurement: kWh
    accuracy_decimals: 3
    state_class: "total_increasing"
    device_class: "energy"
  - name: "Cumulative Reactive Import"
    unit_of_measurement: kvarh
    accuracy_decimals: 3
  - name: "Cumulative Reactive Export"
    unit_of_measurement: kvarh
    accuracy_decimals: 3
  - name: "Momentary Active Import"
    unit_of_measurement: kW
    accuracy_decimals: 3
    state_class: "measurement"
    device_class: "power"
  - name: "Momentary Active Export"
    unit_of_measurement: kW
    accuracy_decimals: 3
    state_class: "measurement"
    device_class: "power"
  - name: "Momentary Reactive Import"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Export"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Active Import Phase 1"
    unit_of_measurement: kW
    accuracy_decimals: 3
    state_class: "measurement"
    device_class: "power"
  - name: "Momentary Active Export Phase 1"
    unit_of_measurement: kW
    accuracy_decimals: 3
    state_class: "measurement"
    device_class: "power"
  - name: "Momentary Active Import Phase 2"
    unit_of_measurement: kW
    accuracy_decimals: 3
    state_class: "measurement"
    device_class: "power"
  - name: "Momentary Active Export Phase 2"
    unit_of_measurement: kW
    accuracy_decimals: 3
    state_class: "measurement"
    device_class: "power"
  - name: "Momentary Active Import Phase 3"
    unit_of_measurement: kW
    accuracy_decimals: 3
    state_class: "measurement"
    device_class: "power"
  - name: "Momentary Active Export Phase 3"
    unit_of_measurement: kW
    accuracy_decimals: 3
    state_class: "measurement"
    device_class: "power"
  - name: "Momentary Reactive Import Phase 1"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Export Phase 1"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Import Phase 2"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Export Phase 2"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Import Phase 3"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Momentary Reactive Export Phase 3"
    unit_of_measurement: kvar
    accuracy_decimals: 3
  - name: "Voltage Phase 1"
    unit_of_measurement: V
    accuracy_decimals: 3
    state_class: "measurement"
    device_class: "voltage"
  - name: "Voltage Phase 2"
    unit_of_measurement: V
    accuracy_decimals: 3
    state_class: "measurement"
    device_class: "voltage"
  - name: "Voltage Phase 3"
    unit_of_measurement: V
    accuracy_decimals: 3
    state_class: "measurement"
    device_class: "voltage"
  - name: "Current Phase 1"
    unit_of_measurement: A
    accuracy_decimals: 3
    state_class: "measurement"
    device_class: "current"
  - name: "Current Phase 2"
    unit_of_measurement: A
    accuracy_decimals: 3
    state_class: "measurement"
    device_class: "current"
  - name: "Current Phase 3"
    unit_of_measurement: A
    accuracy_decimals: 3
    state_class: "measurement"
    device_class: "current"

```

Megamannen, 2023-06-19:
Are all Aidons using HDLC? Have you tried without ?

guni12, 2023-11-24:
Hi @erikfluff, we have exactly the same problem as you with Aidon at Oskarshamn. Getting the same logs as you do from slimmelezer-wt32-eth01. Did you manage to solve the problem?

erikfluff, 2023-11-24:
No, i got an easee ev charger and get info from the meter trough the easee
load balancer now

guni12, 2023-11-24:
OK. Will search on. Thank you for your quick reply.

erikfluff, 2023-11-24:
Let me know IF you find anything