psvanstrom/esphome-p1reader #78: CRC Pass=no after a while

makkanftw, 2023-08-18:
Hi!
Recently built the P1 reader using the PCB from Naesstrom and using this repo and it works fine for a while but then I start getting CRC PASS = NO and then garbled data and after that nothing.
I have [S34U18 (Sanxing SX631)](https://www.vattenfalleldistribution.se/matarbyte/nya-elmataren/) / Vattenfall with P1 active and using D1 mini MCU. Any suggestions?

Logs:
`[11:52:22][I][crc:281]: Telegram read. CRC: F122 = F122. PASS = YES
[11:52:22][D][sensor:093]: 'Cumulative Active Import': Sending state 1626.19897 kWh with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Cumulative Active Export': Sending state 0.00000 kWh with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Cumulative Reactive Import': Sending state 2.60200 kvarh with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Cumulative Reactive Export': Sending state 1235.21204 kvarh with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Momentary Active Import': Sending state 0.76800 kW with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Momentary Active Export': Sending state 0.00000 kW with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Momentary Reactive Import': Sending state 0.00000 kvar with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Momentary Reactive Export': Sending state 0.49600 kvar with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Momentary Active Import Phase 1': Sending state 0.35000 kW with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Momentary Active Export Phase 1': Sending state 0.00000 kW with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Momentary Active Import Phase 2': Sending state 0.26900 kW with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Momentary Active Export Phase 2': Sending state 0.00000 kW with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Momentary Active Import Phase 3': Sending state 0.14800 kW with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Momentary Active Export Phase 3': Sending state 0.00000 kW with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Momentary Reactive Import Phase 1': Sending state 0.00000 kvar with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Momentary Reactive Export Phase 1': Sending state 0.11900 kvar with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Momentary Reactive Import Phase 2': Sending state 0.00000 kvar with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Momentary Reactive Export Phase 2': Sending state 0.14100 kvar with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Momentary Reactive Import Phase 3': Sending state 0.00000 kvar with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Momentary Reactive Export Phase 3': Sending state 0.23400 kvar with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Voltage Phase 1': Sending state 235.30000 V with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Voltage Phase 2': Sending state 236.39999 V with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Voltage Phase 3': Sending state 235.70000 V with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Current Phase 1': Sending state 1.60000 A with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Current Phase 2': Sending state 1.30000 A with 3 decimals of accuracy
[11:52:22][D][sensor:093]: 'Current Phase 3': Sending state 1.20000 A with 3 decimals of accuracy
[11:52:22][W][component:204]: Component <unknown> took a long time for an operation (0.12 s).
[11:52:22][W][component:205]: Components should block for at most 20-30ms.

[11:52:27][D][data:270]: 1-0:2.8.0(000000.00
[11:52:27][D][data:270]: 1-0:3.8.0(000002.602*kvarh	
[11:52:27][D][data:270]: 1
[11:52:27][D][data:270]: 1-0:1.
[11:52:27][D][data:270]: 10:27,0(0 
[11:52:27][D][data:270]:   :3.7.0(0000.
[11:52:27][D][data:270]: \xb9s\xe5
[11:52:27][D][data:270]:  ( 
[11:52:27][D][data:270]: 
[11:52:27][D][data:270]: 
[11:52:27][D][data:270]: 
[11:52:27][D][data:270]: 
[11:52:27][D][data:270]: 
[11:52:27][D][data:270]: 
[11:52:27][D][data:270]: 
[11:52:27][D][data:270]: I\x8es	\xbb\xb0\xff\xfe{\xed\xff\x9b\xbf\xfb\xb0
[11:52:27][D][data:270]: 
[11:52:27][D][data:270]: 4n\x96\xea\xfc\xff~\xff\xbf\xbb\xee\xbf\xdd\xfb\xbf\xf7 \xff\xff\x83\xf3\xfb\xddò\xcc\xd9\xff\xff\xff\xff\xf7\xb7\xff\xf7\x97\xbf\xf7\xf7\xffw\xbf\xf7\xbf\xff\xbf\xf7\xff\xf7\xb7\xbb

[11:52:27][D][data:270]: ,\x95\xe87D\x81\xb2\xc0%
[11:52:27][D][data:270]: x\xaf\xfb\xf3\xbe\xeb\xf7_\xdf\xff\xfb\xff7\xbe
[11:52:27][D][data:270]: 
[11:52:27][D][data:270]: \x80\xec\xff\xfew\xe6~O\x80
[11:52:27][D][data:270]: 
[11:52:27][D][data:270]: @ @\xd8
[11:52:27][D][data:270]: $8
[11:52:27][D][data:270]: 5D<
[11:52:27][D][data:270]: 
[11:52:27][D][data:270]: 
[11:52:27][D][data:270]: 
[11:52:27][D][data:270]: 
[11:52:27][D][data:270]:  \x88
[11:52:27][D][data:270]: 
[11:52:27][D][data:270]: 
[11:52:28][D][data:270]: w\xf3\xfe\xdc\xdfn \x94
[... 251 lines cut ...]
[11:52:52][D][data:270]: 
[11:52:52][D][data:270]: 
[11:52:52][D][data:270]: 
[11:52:52][D][data:270]: 
[11:52:52][D][data:270]: 
[11:52:52][D][data:270]: 
[11:52:52][D][data:270]: 
[11:52:52][D][data:270]: 
[11:52:52][D][data:270]: 
[11:52:52][D][data:270]: 
[11:52:53][D][data:270]: 
[11:52:53][W][component:204]: Component <unknown> took a long time for an operation (1.12 s).
[11:52:53][W][component:205]: Components should block for at most 20-30ms.
[11:52:57][D][data:270]: 
[11:52:57][D][data:270]: 
[11:52:57][D][data:270]: 
[11:52:57][D][data:270]: 
[11:52:57][D][data:270]: 
[11:52:57][D][data:270]: 
[11:52:57][D][data:270]: 
[11:52:57][D][data:270]: 
[11:52:57][D][data:270]: 
[11:52:57][D][data:270]: 
[11:52:57][D][data:270]: 
[11:52:57][D][data:270]: `

Beaky2000, 2023-08-21:
I believe that the  S34U18 (Sanxing SX631) sometimes and for unknown reasons take very long (several seconds) to transmit an update, and it seems that this reader does not work well in those cases.

It looks like there is a pull-request that will adress this so it could be that your problem will be fixed when/if that PR is approved.

You could also try my fork of this project, which I believe works well with the S34U18: https://github.com/Beaky2000/esphome-p1mini

makkanftw, 2023-08-21:
@Beaky2000 I found your fork after giving up troubleshooting this one and has been working perfect for a couple of days now. Thanks!