psvanstrom/esphome-p1reader #80: Update interval

HenkeGoransson, 2023-09-12:
I have a weird issue.
My values updates every second. I believe it used to update every 10 seconds before?
I been running this since without change for 2,5years.

Is there a way to change update intervall?

Beaky2000, 2023-09-12:
I have noticed the same thing with my Sagemcom T211 / Ellevio meter. It used to update every 10 seconds but now updates every second if RTS is constantly high (which it is with this project). This change is in the meter.

I have made a fork of this project, which lets you select update frequency by connecting the RTS signal to a GPIO, which seems to work well with most, but not all, meters. Obviously you would need to modify your ESP module for this to work: https://github.com/Beaky2000/esphome-p1mini

HenkeGoransson, 2023-09-12:
Thanks alot!

jagheterfredrik, 2023-10-22:
I've noticed the same behavior with my Ellevio smart meter, I updated the YAML and let esphome do the filtering, removing the need to modify the hardware setup:
```
    - name: "Cumulative Active Import"
      filters:
       - throttle: 9.5s
      unit_of_measurement: kWh
      accuracy_decimals: 3
      state_class: "total_increasing"
      device_class: "energy"
```