psvanstrom/esphome-p1reader #84: Custom components are deprecated

albinmedoc, 2023-11-02:
There is a deprecation message shown on [ESPHome docs for custom sensor components](https://esphome.io/components/sensor/custom). See below:

> Custom components are deprecated, not recommended for new configurations and will be removed from ESPHome in a future release. Please look at creating a real ESPHome component and “importing” it into your configuration with [External Components](https://esphome.io/components/external_components). You can find some basic documentation on creating your own components at [Contributing to ESPHome](https://esphome.io/guides/contributing#contributing-to-esphome).

> While I do try to keep the ESPHome configuration options as stable as possible and back-port them, the ESPHome API is less stable. If something in the APIs needs to be changed in order for something else to work, I will do so.

This project has to adapt to keep make sure it keeps working through updates.

tubalainen, 2023-11-02:
Projects comes and goes.
This is a fork that is slightly more updated: https://github.com/Beaky2000/esphome-p1mini

albinmedoc, 2023-11-02:
> Projects comes and goes.
> 
> This is a fork that is slightly more updated: https://github.com/Beaky2000/esphome-p1mini

What I can see, that project is also using custom component.

juahan, 2024-02-03:
At least now there is a test branch for getting rid of the custom component.