psvanstrom/esphome-p1reader #99: Current is NOT displayed as negative value

pstrand1959, 2024-11-06:
Negative current values are not correct. 

-1.0A is showed as -2999,000 A

Current Phase 1   2,600 A                  2,6A
Current Phase 2   −2 998,800 A       -1,2A
Current Phase 3   −2 999,100 A       -0,9A

pstrand1959, 2024-11-07:
Made a small fix for this because smallatof doesn't negative values.

    void parseRow(const char* obisCode, const char* value) {
        double obisValue;
        char tmp[16];
        strcpy(tmp, value);
        if (tmp[0] == '-')
        {
          tmp[0] = '0';
          obisValue = -simpleatof(tmp);
        }
        else
        {
          obisValue = simpleatof(tmp);
        }

        parseRow(obisCode, obisValue);
    }

cadwal, 2025-03-02:
Fixed in the current code.