psvanstrom/esphome-p1reader #100: Problem with  ESPHome 2025.2.0

putteg, 2025-02-19:
Seems like ESPHOME has changed syntax starting with  ESPHome 2025.2.0.
platform: ESP8266 is not working anymore

penapena, 2025-02-19:
Also the custom component doesn't work anymore with this version

boozeman, 2025-02-19:
That platform problem can solved just do what the error says. Remove platform and board from esphome: section and add

esp8266:
  board: d1_mini
  framework:
    version: recommended

https://esphome.io/components/esp8266

But that custom component is the tricky one here. Maybe internal DSMR component? https://esphome.io/components/sensor/dsmr.html

Beaky2000, 2025-02-19:
> But that custom component is the tricky one here. 

I have a fork of this project which is still working with the latest ESPHome. Instructions are focused on  the D1 mini ESP module, but it should work with most boards as long as you figure out which pins to use for what:

https://github.com/Beaky2000/esphome-p1mini

putteg, 2025-02-20:
> I have a fork of this project which is still working with the latest ESPHome. Instructions are focused on the D1 mini ESP module, but it should work with most boards as long as you figure out which [...]

Yay, I ended up using your fork anyway :)
Thanks!

closed 2025-02-20 by putteg