psvanstrom/esphome-p1reader #101: Problem with Custom component

tsthst, 2025-02-23:
Have been using your Yaml code to read from my Landis&Gyr E360, and have been working perfect :-) 
(And still does but cannot update to latest version)

I got this error when updating to newest version of ESPHome:

INFO ESPHome 2025.2.0
INFO Reading configuration /config/esphome/elmaaler.yaml...
Failed config

sensor.custom: [source /config/esphome/elmaaler.yaml:48]
  
  The "custom" component has been removed. Consider conversion to an external component.
  https://esphome.io/guides/contributing#a-note-about-custom-components.
  platform: custom
  lambda: |-
    auto meter_sensor = new P1Reader(id(uart_bus));
    App.register_component(meter_sensor);
    return {
      meter_sensor->cumulativeActiveImport,
      meter_sensor->cumulativeActiveExport,
      meter_sensor->cumulativeReactiveImport,
      meter_sensor->cumulativeReactiveExport,
      meter_sensor->momentaryActiveImport,
      meter_sensor->momentaryActiveExport,

Hope it can be solved, og I could get some hints what to do :-)

cadwal, 2025-02-23:
https://github.com/psvanstrom/esphome-p1reader/pull/102

tsthst, 2025-02-23:
Sorry will close this one again.

tsthst, 2025-02-23:
Duplicate