psvanstrom/esphome-p1reader #104: Home Assistant warning use var instead of kvar for reactive_power

skrynklarn, 2025-04-05:
I am encountering a warning in Home Assistant for all reactive import/export sensors.

```
Entity sensor.momentary_reactive_import (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'kVAR' which is not a valid unit for the device class ('reactive_power') it is using; expected one of ['var'];
Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
```

**Request for Update:**
Could we update the code for the ESPHome integration to use var instead of kVAR as the unit of measurement for reactive power sensors? This would align with the requirements of the reactive_power device class.

I am not comfortable making changes to the code directly and submitting a pull request myself, but I wanted to bring this to your attention.

**Workaround:**
As a temporary solution, I have added the following configuration to all the affected reactive import/export sensors, and this resolves the issue:

```yaml
filters:
  - multiply: 1000
unit_of_measurement: "var"
```

This workaround converts the values from kVAR to VAR by multiplying them by 1000 and sets the correct unit of measurement. However, a proper fix in the ESPHome custom component would be ideal.

Thank in advance!

skrynklarn, 2025-04-09:
I manage to get approved in Home Assistant to add kVAR as a unit for reactive power here in this PR https://github.com/home-assistant/core/pull/142558 so this is not needed anymore.

closed 2025-04-09 by skrynklarn

TimoJJ, 2025-05-09:
@skrynklarn  Was your change already merged to HA release?  I'm still getting the kVAR error with HA 2025.5.0 
Looks like HA now expects 'var', 'kvar' so I think your change is there. 
So is this upper case error? Should p1-reader code output kvar units, not kVAR? So should I just change unit of measurement of those sensors to kvar?